zephyrproject-rtos/zephyr #8286: LwM2M: Observe of not allowed value still creates observer

therealprof, 2018-06-08:
The request to observe 3/0/12 returns a 4.05 Method not allowed, yet an observer is added continuously spitting out errors:

> [lib/lwm2m_engine] [DBG] handle_request: No accept option given. Assume OMA TLV.
[lib/lwm2m_engine] [DBG] engine_add_observer: OBSERVER ADDED 3/0/12(3) token:'686d' addr:212.18.24.70
[lib/lwm2m_engine] [DBG] generate_notify_message: [AUTO] NOTIFY MSG START: 3/0/12(3) token:'686d' [212.18.24.70] 55170
[lib/lwm2m_engine] [ERR] generate_notify_message: error in multi-format read (err:-1)

mike-scott, 2018-06-08:
This is totally a bug I can test/fix.  Thank you for finding it.

Also, we need to track adding a server-wide exponential backoff for bad communication so that we don't spam traffic outbound faster than it can be retried/expired (run out of buffers / pending objects).

nashif, 2018-08-26:
@mike-scott ping

closed 2018-08-28 by nashif