zephyrproject-rtos/zephyr #8287: LwM2M: Cancelling an observation doesn't work

therealprof, 2018-06-08:
When I'm observing an object and cancelling the observation I get an error (-2) and the notifications continue to be sent (yeah, never mind our ACK code, I've just implement CONfirmable notifications since Zephyr seems to be the only implementation **not** using NONconfirmable messages ;) ):

> lib/lwm2m_engine] [DBG] handle_request: No accept option given. Assume OMA TLV.
[lib/lwm2m_engine] [DBG] engine_add_observer: OBSERVER ADDED 3/0/13(3) token:'4579' addr:212.18.24.70
[lib/lwm2m_engine] [DBG] lwm2m_notify_observer: NOTIFY EVENT 3/0/13
[lib/lwm2m_engine] [DBG] generate_notify_message: [MANUAL] NOTIFY MSG START: 3/0/13(3) token:'4579' [212.18.24.70] 67162
[lib/lwm2m_engine] [DBG] generate_notify_message: NOTIFY MSG: SENT
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: checking for reply from [212.18.24.70]
[lib/lwm2m_engine] [DBG] notify_message_reply_cb: NOTIFY ACK type:2 code:0.0 reply_token:'4579'
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: reply 0x200000bc handled and removed
[lib/lwm2m_engine] [DBG] lwm2m_notify_observer: NOTIFY EVENT 3/0/13
[lib/lwm2m_rd_client] [DBG] sm_send_registration: registration sent [212.18.24.70]
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: checking for reply from [212.18.24.70]
[lib/lwm2m_rd_client] [INF] do_update_reply_cb: Update callback (code:2.4)
[lwm2m-client] [DBG] rd_client_event: Registration update complete
[lib/lwm2m_rd_client] [INF] do_update_reply_cb: Update Done
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: reply 0x200000bc handled and removed
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: checking for reply from [212.18.24.70]
[lib/lwm2m_engine] [DBG] handle_request: No accept option given. Assume OMA TLV.
[lib/lwm2m_engine] [ERR] handle_request: remove obserer error: -2
[lib/lwm2m_engine] [DBG] generate_notify_message: [MANUAL] NOTIFY MSG START: 3/0/13(3) token:'4579' [212.18.24.70] 77198
[lib/lwm2m_engine] [DBG] generate_notify_message: NOTIFY MSG: SENT
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: checking for reply from [212.18.24.70]
[lib/lwm2m_engine] [DBG] notify_message_reply_cb: NOTIFY ACK type:2 code:0.0 reply_token:'4579'
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: reply 0x200000bc handled and removed
[lib/lwm2m_engine] [DBG] lwm2m_notify_observer: NOTIFY EVENT 3/0/13
[lib/lwm2m_engine] [DBG] generate_notify_message: [MANUAL] NOTIFY MSG START: 3/0/13(3) token:'4579' [212.18.24.70] 87234

mike-scott, 2018-06-08:
Nice typo there in the error message too.

mike-scott, 2018-06-08:
> yeah, never mind our ACK code, I've just implement CONfirmable notifications since Zephyr seems to be the only implementation not using NONconfirmable messages

We can add another issue for this as well.  But, I'll need to refer to the LwM2M Tech spec to see what it says there.

therealprof, 2018-06-08:
Spec says both are okay which is why I went ahead and implemented it. ;)

nashif, 2018-08-26:
@mike-scott ping

therealprof, 2018-08-27:
Interesting. Let me see whether I can still reproduce it and get packet captures for you.

therealprof, 2018-08-28:
Soooo. We're trying to cancel the observation with an `observe cancel` message, however as it turns out we're not using the right token to do so. Let me try to fix that and retry.

mike-scott, 2018-08-29:
@therealprof did you get a chance to test this again?

therealprof, 2018-08-29:
Yes, works just fine (as usual when you do it right 😀).

mike-scott, 2018-08-29:
Ok, closing this issue.

closed 2018-08-29 by mike-scott