zephyrproject-rtos/zephyr #8299: Failed test: kernel.memory_pool.mpool_alloc_free_isr (in tests/kernel/mem_pool/mem_pool_api)

zephyrbot, 2018-06-09:
Fails on quark_se_c1000_ss_devboard

```

***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
 Running test suite mpool_api
 ===================================================================
 starting test - test_mpool_alloc_free_thread
 PASS - test_mpool_alloc_free_thread
 ===================================================================
 starting test - test_mpool_alloc_free_isr
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40000d9e
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40000d9e
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
[... 3540 lines cut ...]
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x0
 Address 0x40002e9c
 Current thread ID = 0x00000200
 Faulting instruction address = 0x40002e9c
 Exception vector: 0x3, cause code: 0x4, parameter 0x

```

nashif, 2018-06-09:
this seem to be resolved with #8297

nashif, 2018-06-09:
@vonhust fyi

vonhust, 2018-06-09:
It is caused by the bug in irq_offload

We reproduce it in em9d and em7d.  

@nashif  it should be  fixed by #8297

nashif, 2018-06-09:
@vonhust yes, I confirm this get fixed with #8297, actually it solves many issues on ARC, confirming now

closed 2018-06-11 by MaureenHelm