zephyrproject-rtos/zephyr #8300: Failed test: kernel.memory_protection.userspace.access_after_revoke (in tests/kernel/mem_protect/userspace)

zephyrbot, 2018-06-09:
Fails on em_starterkit_em7d_v22

```
***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
 Running test suite userspace
 ===================================================================
 starting test - is_usermode
 PASS - is_usermode
 ===================================================================
 starting test - write_control
 Exception vector: 0x7, cause code: 0x0, parameter 0x0
 Address 0x6ac
 Current thread ID = 0x800081b4
 Faulting instruction address = 0x6ac
 Caught system error -- reason 0
 PASS - write_control
 ===================================================================
 starting test - disable_mmu_mpu
 Exception vector: 0x7, cause code: 0x0, parameter 0x0
 Address 0x2ea4
 Current thread ID = 0x800081b4
 Faulting instruction address = 0x2ea4
 Caught system error -- reason 0
 PASS - disable_mmu_mpu
 ===================================================================
 starting test - read_kernram
 Exception vector: 0x6, cause code: 0x1, parameter 0x4
 Address 0x80000924
 Current thread ID = 0x800081b4
 Faulting instruction address = 0x9de
 Caught system error -- reason 0
 PASS - read_kernram
 ===================================================================
 starting test - write_kernram
 Exception vector: 0x6, cause code: 0x1, parameter 0x4
 Address 0x80000924
 Current thread ID = 0x800081b4
 Faulting instruction address = 0xa3a
 Caught system error -- reason 0
 PASS - write_kernram
 ===================================================================
 starting test - write_kernro
 Exception vector: 0x6, cause code: 0x2, parameter 0x4
 Address 0x9124
 Current thread ID = 0x800081b4
 Faulting instruction address = 0xace
 Caught system error -- reason 0
 PASS - write_kernro
 ===================================================================
 starting test - write_kerntext
 Exception vector: 0x6, cause code: 0x2, parameter 0x4
 Address 0x6510
 Current thread ID = 0x800081b4
 Faulting instruction address = 0x5d4
 Caught system error -- reason 0
 PASS - write_kerntext
 ===================================================================
 starting test - read_kernel_data
 Exception vector: 0x6, cause code: 0x1, parameter 0x4
 Address 0x800080a8
 Current thread ID = 0x800081b4
 Faulting instruction address = 0xb14
 Caught system error -- reason 0
 PASS - read_kernel_data
 ===================================================================
 starting test - write_kernel_data
 Exception vector: 0x6, cause code: 0x2, parameter 0x4
 Address 0x800080a8
 Current thread ID = 0x800081b4
 Faulting instruction address = 0xb72
 Caught system error -- reason 0
 PASS - write_kernel_data
 ===================================================================
 starting test - read_priv_stack
 Exception vector: 0x6, cause code: 0x1, parameter 0x4
 Address 0x800048d8
 Current thread ID = 0x800081b4
 Faulting instruction address = 0xbd4
 Caught system error -- reason 0
 PASS - read_priv_stack
 ===================================================================
 starting test - write_priv_stack
 Exception vector: 0x6, cause code: 0x2, parameter 0x4
 Address 0x800048d8
 Current thread ID = 0x800081b4
 Faulting instruction address = 0xc52
 Caught system error -- reason 0
 PASS - write_priv_stack
 ===================================================================
 starting test - pass_user_object
 0x800001d0 is not a valid k_sem
 syscall _handler_k_sem_init failed check: access deniException vector: 0x6, cause code: 0x2, parameter 0x2
 Address 0x800047fc
 ***** Stack Check Fail! *****
 Current thread ID = 0x800081b4
 Caught system error -- reason 2

     Assertion failed at zephyr.git/tests/kernel/mem_protect/userspace/src/main.c:80: _SysFatalErrorHandler: (Reached unreachable code)
 Unexpected fault during test
 Exception vector: 0xd, cause code: 0x0, parameter 0x0
 Address 0xfffffffd
 Current thread ID = 0x80000894
 Faulting instruction address = 0x380e
 Caught system error -- reason 0

     Assertion failed at zephyr.git/tests/kernel/mem_protect/userspace/src/main.c:80: _SysFatalErrorHandler: (Reached unreachable code)
 Unexpected fault during test
 ASSERTION FAIL [!(thread->base.user_options & (1 << 0))] @ zephyr.git/kernel/thread_abort.c:35:
 	essential thread aborted

```

nashif, 2018-06-09:
CONFIG_PRIVILEGED_STACK_SIZE=512 makes the test pass, not sure this is the right fix though, looks like we are failing in the fatal handler:

`Faulting instruction address = 0x33e6`

is in uart_ns16550_poll_out

vonhust, 2018-06-10:
@nashif   it's as expected.  this exception comes out in 
`Z_OOPS(Z_SYSCALL_OBJ_INIT(sem, K_OBJ_SEM));`
The kernel oops will call printk to output error msg, but the privilege stack (256 bytes) is overflow, and caught by stack check. 

The fix is to raise the privilege stack size to 384 bytes or 512 bytes.

It seems 256 bytes of privilege stack is not enough for ARC when printk is called.

vonhust, 2018-06-11:
Fixed by #8297

PR #8297 (2018-06-11): arch: arc: use top of isr stack as exception stack and bug fixes
* re-use top of isr stack as exception stack
* bug fixes in irq offload's implementation, this bug will cause irq_offload work wrongly.
* improve kernel oops implementation
* adjust the privilege  stack size to 384 bytes to avoid the possible stack overflow caused by printk, interrupt and other cases,  see a91f4e1
* modify the test condition for emsk_em7d_v22 in tests/kernel/sched/schedule_api

This PR may 
  fixes #8292 
  fixes #8300 
  fixes #8090
closed 2018-06-11 by MaureenHelm