zephyrproject-rtos/zephyr #8309: lpcxpresso54114_m4: when i configed system clock from 48M to 96M the target can't work.

Jeepgoing, 2018-06-11:
hi,all:

i am testing lpcxpresso54114, i found when i configed system clock from 48M to 96M the target can't work, i have compare `arch\arm\soc\nxp_lpc\lpc54xxx\soc.c` with the SDK supported by NXP, i found zephyr lost function `POWER_SetVoltageForFreq(96000000U);`, but i can't find `POWER_SetVoltageForFreq` source code in SDK, so anyone else have met this issue?

MaureenHelm, 2018-06-25:
`POWER_SetVoltageForFreq` is defined in a library in the MCUXpresso SDK in `devices/LPC54114/gcc/libpower_cm4_hardabi.a`. There are cm4_softabi and cm0 variants of the library as well. Since these functions are not available in source form, we couldn't import them into Zephyr. This limitation may change after #7516 gets resolved, but for now you can update your project to link in the library from the SDK.

closed 2018-06-25 by MaureenHelm

Jeepgoing, 2018-06-26:
@MaureenHelm thanks, i have solved this issue according to your advise (link libpower_cm4_softabi.a)