zephyrproject-rtos/zephyr #8320: 1.13 Release Checklist

MaureenHelm, 2018-06-11:
Add items as required
--------
- [ ] Major enhancements:
  - [ ] QM (Quality Managed) level qualification: #6906
  - [ ] Support Precision Time Protocol (PTP): #3067
  - [ ] Ecosystem and Tool Support: #2934 
  - [ ] Logging Subsystem Enhancements: #6186
  - [ ] add backtrace support when we have a crash: #4416
  - [ ] Multiple Git Repositories: #6770
  - [ ] Advanced Power Management: #3295 
  - [ ] Bluetooth: Deprecate TinyCrypt and use mbedTLS: #8081 
  - [ ] Add generation of SPDX TagValue documents to each build: #7317

- [ ] Pre RC1 Steps
   - [ ] Verify that all external components and external dependencies are up to date, for example 
      - mbedTLS
      - tinycrypt
      - FatFS
   - [ ] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as unexploitable.

See https://github.com/zephyrproject-rtos/zephyr/wiki/Release-procedure for tagging details

--------
- [ ] Release Notes WIP
  - [ ] Create draft doc/release-notes-1.13.rst with feature summary from https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update doc/release-notes-1.13.rst details sections (relative to 1.12.0 release) as important issues are addressed (subsystem owners)
--------
- [ ] Release data
  - [ ] Update version in VERSION
  - [ ] Run sanitycheck with --all --release and checkin resulting file scripts/sanity_chk/sanity_last_release.csv
--------
- [ ] Finalize Release Notes
  - [ ] Add list of GitHub issues (spell check issue list and fix in GitHub as needed)
  - [ ] Review doc/release-notes-1.13.rst overview summary and details sections (see Sign-off below)
  - [ ] Update doc/release-notes.rst (add reference to 1.13 notes)
  - [ ] Update doc/index.rst (add reference to doc/1.13.0)
  - [ ] Update doc/LICENSING.rst for new components that do not have a license
  - [ ] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases
  - [ ] Final tagged release as zephyr-v1.13.0 (with release notes)
  - [ ] Update https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update https://www.zephyrproject.org/developers/#downloads (and git checkout version example)
  - [ ] Update doc generation (in future should be automatic based on tag)
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release
--------
- [ ]  Final Release Notes Sign-off
  - Subsystems
    - [ ] Networking - @jukkar 
    - [ ] Bluetooth - @jhedberg , @carlescufi 
  - Architectures
    - [ ] ARC - @ruuddw 
    - [ ] ARM - @galak, @MaureenHelm 
    - [ ] X86 - @nashif, @andrewboie 
    - [ ] Xtensa, RISCV32, NIOS-II - @nashif, @andrewboie 
  - [ ] Security - @agross-linaro 
--------
- [ ] After Release
  - [ ] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [ ] Create branch for 1.13.x releases, update github protections for branch
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [ ] Update https://www.zephyrproject.org with latest stats from https://zephyr.biterg.io

pfalcon, 2018-08-20:
Added TLS/DTLS sockets https://github.com/zephyrproject-rtos/zephyr/pull/9007 to major enhancements. @rlubos FYI.

pfalcon, 2018-08-20:
https://github.com/zephyrproject-rtos/zephyr/pull/9391 should be documented as non backward-compatible change.

nashif, 2018-09-07:
> #9391 should be documented as non backward-compatible change.

please add a line about this in the release notes draft.

nashif, 2018-09-10:
@ruuddw, @vonhust  can you check the release notes please and see if ARC/synopsys changes are covered? 

@galak, @MaureenHelm ditto

MaureenHelm, 2018-09-10:
@nashif Arm is done.

ruuddw, 2018-09-10:
@nashif Synopsys/ARC changes are covered, just few suggestions to improve:
Designware -> Synopsys DesignWare 
arch: Support Synopsys nSIM device simulator -> please move (or repear) to Boards/SoC section, and rephrase to "arc: Support Synopsys ARC nSIM instruction set simulator"

nashif, 2018-09-11:
Check list for 1.14 created: #9898

closed 2019-02-21 by nashif