zephyrproject-rtos/zephyr #8329: qustion: build-system: How to generate a preprocess file (*.i)

zhaozhongchen, 2018-06-12:
Hi:

'make help' can print which file supports generate a preprocess file, I want preprocess a specify file (drivers/serial/uart_stm32.c) but that file doesn't has a make command to generate.

```
$ make help | grep uart
... drivers/console/uart_console.obj
... drivers/console/uart_console.i
... drivers/console/uart_console.s
```

How to make the specify file can generate a preprocess file?

SebastianBoe, 2018-06-12:
Usually, I run ninja/make in verbose mode (ninja -v), find the command that compiles the source file I am interested in, and then copy-and-modify it by adding the -E flag. This will produce preprocessed output in the .obj file.

Something more sophisticated would be welcome.

SebastianBoe, 2018-06-12:
I have seen that some source files have a .i target, and some do not, I have never bothered looking into why though.

zhaozhongchen, 2018-06-12:
@SebastianBoe 

some files can generate a proprecess file and some files not. I just curious which CMakeFile control this feature.

```
➜  zephyr git:(rs485) ✗ pwd
/home/qianfan/zephyr/samples/hello_world/build/zephyr
➜  zephyr git:(rs485) ✗ make help
The following are some of the valid targets for this Makefile:
... all (the default if no target is provided)
... clean
... depend
... edit_cache
... run
... linker_pass_final_script
... rebuild_cache
... syscall_list_h_target
... syscall_macros_h_target
... driver_validation_h_target
... linker_script
... zephyr
... kernel_elf
... offsets_h
... zephyr_prebuilt
... kobj_types_h_target
... offsets
... arch/arm/core/offsets/offsets.obj
... arch/arm/core/offsets/offsets.i
... arch/arm/core/offsets/offsets.s
... arch/arm/soc/st_stm32/common/stm32cube_hal.obj
... arch/arm/soc/st_stm32/common/stm32cube_hal.i
... arch/arm/soc/st_stm32/common/stm32cube_hal.s
... arch/arm/soc/st_stm32/stm32f0/soc.obj
... arch/arm/soc/st_stm32/stm32f0/soc.i
... arch/arm/soc/st_stm32/stm32f0/soc.s
... arch/arm/soc/st_stm32/stm32f0/soc_gpio.obj
... arch/arm/soc/st_stm32/stm32f0/soc_gpio.i
... arch/arm/soc/st_stm32/stm32f0/soc_gpio.s
... arch/common/isr_tables.obj
... arch/common/isr_tables.i
... arch/common/isr_tables.s
... drivers/clock_control/stm32_ll_clock.obj
... drivers/clock_control/stm32_ll_clock.i
... drivers/clock_control/stm32_ll_clock.s
... drivers/clock_control/stm32f0x_ll_clock.obj
... drivers/clock_control/stm32f0x_ll_clock.i
... drivers/clock_control/stm32f0x_ll_clock.s
... drivers/console/uart_console.obj
... drivers/console/uart_console.i
... drivers/console/uart_console.s
... drivers/interrupt_controller/exti_stm32.obj
... drivers/interrupt_controller/exti_stm32.i
... drivers/interrupt_controller/exti_stm32.s
... drivers/pinmux/stm32/pinmux_stm32.obj
... drivers/pinmux/stm32/pinmux_stm32.i
... drivers/pinmux/stm32/pinmux_stm32.s
... drivers/timer/cortex_m_systick.obj
... drivers/timer/cortex_m_systick.i
... drivers/timer/cortex_m_systick.s
... drivers/timer/sys_clock_init.obj
... drivers/timer/sys_clock_init.i
... drivers/timer/sys_clock_init.s
... ext/hal/st/stm32cube/stm32f0xx/drivers/src/stm32f0xx_hal.obj
... ext/hal/st/stm32cube/stm32f0xx/drivers/src/stm32f0xx_hal.i
... ext/hal/st/stm32cube/stm32f0xx/drivers/src/stm32f0xx_hal.s
... ext/hal/st/stm32cube/stm32f0xx/drivers/src/stm32f0xx_hal_rcc.obj
... ext/hal/st/stm32cube/stm32f0xx/drivers/src/stm32f0xx_hal_rcc.i
... ext/hal/st/stm32cube/stm32f0xx/drivers/src/stm32f0xx_hal_rcc.s
... ext/hal/st/stm32cube/stm32f0xx/drivers/src/stm32f0xx_ll_utils.obj
... ext/hal/st/stm32cube/stm32f0xx/drivers/src/stm32f0xx_ll_utils.i
... ext/hal/st/stm32cube/stm32f0xx/drivers/src/stm32f0xx_ll_utils.s
... ext/hal/st/stm32cube/stm32f0xx/soc/system_stm32f0xx.obj
... ext/hal/st/stm32cube/stm32f0xx/soc/system_stm32f0xx.i
... ext/hal/st/stm32cube/stm32f0xx/soc/system_stm32f0xx.s
... isr_tables.obj
... isr_tables.i
... isr_tables.s
... lib/crc/crc16_sw.obj
... lib/crc/crc16_sw.i
... lib/crc/crc16_sw.s
... lib/crc/crc8_sw.obj
... lib/crc/crc8_sw.i
... lib/crc/crc8_sw.s
... lib/mempool/mempool.obj
... lib/mempool/mempool.i
... lib/mempool/mempool.s
... lib/rbtree/rb.obj
... lib/rbtree/rb.i
... lib/rbtree/rb.s
... lib/thread_entry.obj
... lib/thread_entry.i
... lib/thread_entry.s
... misc/empty_file.obj
... misc/empty_file.i
... misc/empty_file.s
... misc/generated/configs.obj
... misc/generated/configs.i
... misc/generated/configs.s
... misc/printk.obj
... misc/printk.i
... misc/printk.s
```

SebastianBoe, 2018-06-12:
"I just curious which CMakeFile control this feature."

You and me both, this behaviour is not intentional :)

zhaozhongchen, 2018-06-12:
@SebastianBoe 
> Usually, I run ninja/make in verbose mode (ninja -v)

can make supports verbose mode, like ninja -v?
 'make V=1' doesn't works.

SebastianBoe, 2018-06-12:
Try "make VERBOSE=1"

zhaozhongchen, 2018-06-12:
@SebastianBoe Thanks!

closed 2018-08-15 by SebastianBoe