zephyrproject-rtos/zephyr #8331: Unify counter.h and rtc.h interfaces

nordic-krch, 2018-06-12:
Counter.h and rtc.h share almost the same functionality with only minor
differences:
- rtc.h has option for initial value and initial alarm (set by rtc_set_config)
- counter.h has pointer to the user context in the alarm callback

Both interfaces have drawbacks:
- no information about the frequency so alarm is set in ticks which are virtual
- no helper function to convert microseconds or milliseconds to ticks
- in both APIs it is unclear if alarm is one shot or periodic
- interfaces are limiting functionality of peripherals (TIMER, RTC) which usually
  have couple of capture/compare channels. Current interfaces allow usage of single channel
- lack of information about overflow (unless alarm is periodic and is actual overflow)
- lack of possibility to set alarm for absolute value

Counter.h and RTC.h should be unified. Implementation of unified interface can be based on low power RTC-like peripheral or more accurate high frequency general purpose timers.

Topic already mentioned in #5895

carlescufi, 2018-06-12:
Related issues: https://github.com/zephyrproject-rtos/zephyr/issues/6498 and https://github.com/zephyrproject-rtos/zephyr/issues/7520

carlescufi, 2019-02-12:
Merged into master

closed 2019-02-12 by carlescufi