zephyrproject-rtos/zephyr #8334: nrf52840.dtsi contains "0x" in device label

barsok, 2018-06-12:
USBD has description:
>usbd: usbd@**0x**40027000 {

which causes a warning during compilation of the devicetree

> -- Generating zephyr/include/generated/generated_dts_board.h
>nrf52840_pca10056.dts_compiled: Warning (unit_address_format): Node /soc/usbd@0x40027000 unit name should not have leading "0x"
>extract_dts_includes.py: Merge of 'title': 'Nordic nRF52 USBD'  overwrites 'USB Base Structure'.
>extract_dts_includes.py: Merge of 'node_type': '['USB-EP', 'USB']'  overwrites '['USB']'.
>extract_dts_includes.py: Merge of 'description': 'This binding gives a base representation of the Nordic >nRF52 USB device controller

and strangely looking labels in generated_dts_board.h:

> #define NORDIC_NRF_USBD_0X40027000_BASE_ADDRESS                 **0x**40027000

barsok, 2018-06-12:
BTW the fixup file for nrf52840 also needs to be amended as it contains defines with the 0x

alext-mkrs, 2018-06-16:
I take this one if no one objects.

PR #8433 (2018-06-16): dt: nrf52840: remove 0x from USBD address
The prepended 0x causes DTC warnings and shouldn't be there.

Tested by compiling hello_world for nrf52840_pca10056 before and after.

Fixes #8334.
commit 0824ec64 (2018-06-16): dt: nrf52840: remove 0x from USBD address

The prepended 0x causes DTC warnings and shouldn't be there.

Tested by compiling hello_world for nrf52840_pca10056 before and after.

Fixes #8334.
closed 2018-06-16 by carlescufi