zephyrproject-rtos/zephyr #8355: CMake prints a spammy warning about "policy CMP0000"

ulfalizer, 2018-06-13:
I'm a CMake noob, so I have no idea what the proper fix is here, but having 12 lines of spam about CMP0000 whenever you run CMake is pretty bad:

```
CMake Deprecation Warning at /home/ulf/zephyr/cmake/app/boilerplate.cmake:38 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)
```

SebastianBoe, 2018-06-13:
This affects CMake version >= 3.9.0 so the simplest solution is to use CMake 3.8.x.

Last I looked into this I concluded that longterm we need to either fork CMake or convince upstream to allow projects to suppress this warning, as they were allowed to before 3.9.x.

Any other solution suggestions are welcome.

ulfalizer, 2018-06-13:
What about not using "the OLD behavior for policy CMP0000", whatever that means?

SebastianBoe, 2018-06-13:
IIRC that creates bigger problems than this warning, can't remember the details though.

ulfalizer, 2018-06-13:
Probably worth looking into.

ulfalizer, 2018-06-19:
@tejlmand 
Do you know anything about this btw? :)

SebastianBoe, 2018-06-19:
" but I could imagine that it makes life harder if bumping the CMake version in future as all files would need to be updated"

Yep. And more importantly, backwards-compatibility with customer applications.

SebastianBoe, 2018-06-19:
I wonder, does CMake support multiple invocations of minimum required?

Then we would not have a compatibility issue when bumping the version ...

SebastianBoe, 2018-06-19:
I imagine they don't, the whole point is for them to be able to determine how CMake should behave as early as possible ...

ulfalizer, 2018-06-19:
Guess it might need to be changed anyway eventually, since the warning says that the behavior will be removed.

tejlmand, 2018-06-20:
Multiple invocations of cmake_minimum_required is allowed.
This is very useful when you include other projects, which might have different requirements.

Imagine if you have project A, e.g. Zephyr, which requires version 3.8, and you then include mbedTLS, which requires, let's say version 3.10 in its CMakeLists.txt file.
Then CMake will complain if you're running version 3.9.

But versions 3.10 and above are fine, as it satisfies the versions specified for both Zephyr and mbedTLS :)

tejlmand, 2018-06-20:
@SebastianBoe Do you know why version 3.8.2 is chosen ?
Is zephyr using any 3.8 specific features or was it just the version most people were running at that time, so it's not verified to work with older versions. (which is a very good reason)

For fun, I just tried run CMake version 3.5.1 without any apparent problems (except I had to specify a lower required version)

https://cmake.org/cmake/help/v3.8/release/3.8.html

SebastianBoe, 2018-06-20:
> Multiple invocations of cmake_minimum_required is allowed.

Great!

Then it is safe to duplicate the version information in every application AFAICT.

I can't recall how we landed on 3.8.2, you would have to check the git history, there has been a back-and-forth. If the version is too high it won't be packaged by distrubutions. If the version is too low you will lose out on features and bugfixes.

It might have been that we wanted to support 3.8.2 so that users could avoid the issue reported here. But since multiple invocations of cmake_minimum_required is permitted we might not need to work around it any more.

henrikbrixandersen, 2018-06-21:
Still, `git ls-files '**/CMakeLists.txt' | xargs sed -i '1s/^/cmake_minimum_required(VERSION 3.8.2)\n/'` gives us approximately 780 added lines of boiler plate code...

tejlmand, 2018-06-22:
you should disregard files under subsys/ ext/ kernel/, etc.
Only the entry point for the applications, i.e. all CMakeLists.txt files under  samples/ and tests/.
which still amounts to roughly ~150 under samples and ~200 under tests.

Basically all the files which contains the line project(none) would most likely also need the line cmake_minimum_required()

SebastianBoe, 2018-06-22:
@henrikbrixandersen : Yes, not great, but for those that use CMake version > 3.8.x I believe this is the best solution (apart from convincing upstream to re-introduce support for having cmake_minimum_required outside of the toplevel CMakeLists.txt).

PR #9186 (2018-08-15): Fix policy 000 warning
Prepend the text 'cmake_minimum_required(VERSION 2.4)' into the
application and test build scripts.

Modern versions of CMake will spam users with a deprecation warning
when the toplevel CMakeLists.txt does not specify a CMake
version. This is documented in bug #8355.

To resolve this we include a dummy cmake_minimum_required() line into
the toplevel build scripts. This version is then overridden by
boileplate.cmake to avoid duplicating the true minimum required
version between all build scripts.

Also change CMake policy CMP000 from OLD to NEW which in turn
finally rids [... cut]
commit 6682652a (2018-08-15): cmake: change policy CMP0000 to NEW

Change CMake policy CMP0000 from OLD to NEW to resolve #8355.
closed 2018-08-15 by nashif