zephyrproject-rtos/zephyr #8356: Failed test: kernel.common.atomic (tests/kernel/common) on Altera Max10

spoorthik, 2018-06-13:
**Latest commit on which the test failed:** 72959543991917ea09d47cdb5c0b941e8947c6fc

**Execution sample log:**
```
***** Booting Zephyr OS v1.12.0-110-g7295954 (delayed boot 1000ms) *****
Running test suite common

===================================================================

starting test - test_byteorder_memcpy_swap

PASS - test_byteorder_memcpy_swap

===================================================================

starting test - test_byteorder_mem_swap

PASS - test_byteorder_mem_swap

===================================================================

starting test - test_atomic

PASS - test_atomic

===================================================================

starting test - test_bitfield

twiddling bits....

    Assertion failed at zephyr.git/tests/kernel/common/src/bitfield.c:30: test_bitfield

sys_set_bit failed on bit 0
FAIL - test_bitfield

===================================================================

starting test - test_printk

1 4294967295 ffffffff22 113 10000 32768 40000 22
p 112 -10000 -32768 -40000 -22
0xcafebabe 0x0000beef
0x1 0x01 0x0001 0x00000001
0x1 0x 1 0x   1 0x       1
42 42 0042 00000042
-42 -42 -042 -0000042
42 42   42       42
42 42 0042 00000042
255     42    abcdef  0x0000002a      42
-1 4294967295 ffffffff
PASS - test_printk

===================================================================

starting test - test_slist

PASS - test_slist

===================================================================

starting test - test_dlist
PASS - test_dlist

===================================================================

starting test - test_intmath

PASS - test_intmath

===================================================================

starting test - test_timeout_order

thread 0 synced on timer 0
thread 1 synced on timer 1
thread 2 synced on timer 2
PASS - test_timeout_order

===================================================================

starting test - test_clock_uptime

PASS - test_clock_uptime

===================================================================

starting test - test_clock_cycle

PASS - test_clock_cycle

===================================================================

starting test - test_version

PASS - test_version

===================================================================

starting test - test_multilib

PASS - test_multilib

===================================================================

===================================================================

PROJECT EXECUTION FAILED
```

pswarnak, 2018-06-13:
@spoorthik 
Possible duplication of https://github.com/zephyrproject-rtos/zephyr/issues/7909
Please check.

nashif, 2018-06-13:
it is a duplicate

closed 2018-06-13 by spoorthik