zephyrproject-rtos/zephyr #8357: bluetooth: request the capability to change gap device name programmatically

jli157, 2018-06-13:
The device name of a bluetooth device is currently configured by the macro `CONFIG_BT_DEVICE_NAME`, which presents the device from being named by the device name data stored in its internal flash memory or OTP memory.  So, an API shall be added to support changing device name programmatically.

carlescufi, 2018-06-13:
I am all in favor of this. I do not see why we couldn't allow this, barring current technical limitations on the way we handle advertising data.

jhedberg, 2018-06-13:
You can already now do this for the name that's present in advertising data, however for the name that's exposed through the GAP GATT service we'd need to add a new API. It should be fairly simple to do, however since it's currently just `static const char *gap_name = CONFIG_BT_DEVICE_NAME;` in `subsys/bluetooth/host/gatt.c`. I.e. we need a function to assign some other string to it.

Vudentz, 2018-06-15:
I guess we would also need to persist it as well and perhaps have some means to notify about name changes so it can update adv data.

PR #8848 (2018-07-23): Bluetooth: Add support for changing Device Name
This adds support for changing GAP Device Name at runtime and persists it in a flash if BT_SETTINGS is defined.

Note: It is also possible to set BT_DEVICE_NAME_STORAGE without BT_SETTINGS but that will be overwritten in the next boot thus why there is a line with default 0 if !BT_SETTINGS.

Fixes #8357
commit 2637c978 (2018-07-23): Bluetooth: Store device name

This uses bt_dev to store the name and allow changing it at runtime, in
addtion to that if CONFIG_BT_SETTINGS is defined make the name
persistent.

Fixes #8357
commit aa339ed0 (2018-07-23): Bluetooth: GATT: Make GAP name writable

This makes GAP name writable if CONFIG_BT_DEVICE_NAME_STORAGE is > 0
which means the name can be persisted.

Fixes #8357
closed 2018-07-23 by jhedberg