zephyrproject-rtos/zephyr #8360: extract_dts_includes.py should not be triggering warnings

SebastianBoe, 2018-06-13:
In the latest release extract_dts_includes.py has started issuing warnings for Zephyr samples.

Example:

```
sebo@mach:~/zephyr/samples/hello_world$ cmake -H. -Boee -DBOARD=nrf52840_pca10056
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD nrf52840_pca10056
Zephyr version: 1.12.99
Parsing Kconfig tree in /home/sebo/zephyr/Kconfig
Using /home/sebo/zephyr/boards/arm/nrf52840_pca10056/nrf52840_pca10056_defconfig as base
Merging /home/sebo/zephyr/samples/hello_world/prj.conf
-- Generating zephyr/include/generated/generated_dts_board.h
extract_dts_includes.py: Merge of 'title': 'Nordic nRF52 USBD'  overwrites 'USB Base Structure'.
extract_dts_includes.py: Merge of 'description': 'This binding gives a base representation of the Nordic nRF52 USB device controller
'  overwrites 'This binding gives the base structures for all USB devices
'.
extract_dts_includes.py: Merge of 'node_type': '['USB-EP', 'USB']'  overwrites '['USB']'.
-- Cache files will be written to: /home/sebo/.cache/zephyr
-- The C compiler identification is GNU 6.2.0

```

Zephyr samples should be free of warnings.

I would suggest to rectify this by turning all warnings into errors in CI and then analyze all the CI warnings and fix all of them by either resolving or suppressing each warning.

In addition to turning warnings into errors in CI, one might consider turning some of the warnings into errors on-the-desk as well.

Relates to: https://github.com/zephyrproject-rtos/zephyr/issues/7744

b0661, 2018-06-16:
This is a bug in extract_dts_includes.py. Sample is ok.

pabigot, 2018-12-26:
This has certainly not been fixed; in current master it produces the following when building `hello_word` for `frdm_k64f`:
```
Configuration written to '/mnt/devel/external/zephyr/samples/hello_world/k64f/zephyr/.config'
-- Loading /mnt/devel/external/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts as base
-- Overlaying /mnt/devel/external/zephyr/dts/common/common.dts
extract_dts_includes.py: Merge of 'category': 'required'  overwrites 'optional'.
extract_dts_includes.py: Merge of 'category': 'required'  overwrites 'optional'.
-- Cache files will be written to: /home/pab/.cache/zephyr
...
```
By adding more information to the diagnostic I've determined these come from `nxp,kinetis-uart.yaml` and `nxp,kinetis-dspi.yaml` since both override the generic node to make the `clock` property required.  The originally reported diagnostic for `nrf52840_pca10056` has been fixed some other way.

This sort of opaque diagnostic of unclear importance creates a negative impression of Zephyr's overall quality when presented to users who are just trying to build a works-everywhere sample application.

pabigot, 2019-01-13:
The case I'd mentioned has been improved by #12383, producing:
```
-- Overlaying /mnt/devel/external/zephyr/dts/common/common.dts
extract_dts_includes.py: /mnt/devel/external/zephyr/dts/bindings/serial/nxp,kinetis-uart.yaml('clocks') merge of property 'category': 'required' overwrites 'optional'.
extract_dts_includes.py: /mnt/devel/external/zephyr/dts/bindings/spi/nxp,kinetis-dspi.yaml('clocks') merge of property 'category': 'required' overwrites 'optional'.
```
However, the intent of this issue as I understand it is that even these warnings should not be produced on samples; the issue should remain open.

galak, 2019-01-31:
Closing this since we don't seem to get any warnings right now (looking at hello world on all boards).  Let's re-open an issue if this comes up again.

nashif, 2020-08-11:
that script does not exist anymore, obsolete.

PR #8435 (2018-07-04): scripts: extract_dts_includes.py: fix false info message
Do not provide info message if node type, title, version,
node description is overwritten.
This is the expected behaviour.

Fixes #8360
commit ca7fc2ad (2018-07-04): scripts: extract_dts_includes.py: fix false info message

Do not provide info message if node type, title, version,
node description is overwritten.
This is the expected behaviour.

Fixes #8360
closed 2020-08-12 by nashif