zephyrproject-rtos/zephyr #8364: mcumgr: unable to properly read big files

nvlsianpu, 2018-06-13:
Can't to read bigger files. Looks lik it is possible to write big files (tested up to 8 KB) - they are sends as small chunks. Reading of such files is impossible - suspected thet the library can not divide bigger files into chunks.

Issue reported for the smp_svr@nrf52_pca10040

carlescufi, 2018-06-26:
CC @philips77

nashif, 2018-09-04:
external project issue, lowering priority.

rljordan, 2019-03-25:
Any updates on this issue?

nvlsianpu, 2020-02-13:
cc @de-nordic

de-nordic, 2020-02-14:
My first finding is that: the problem is with cbor encoder running out of memory, while trying to encode  single buffer of  FS_MGMT_DL_CHUNK_SIZE.

de-nordic, 2020-02-14:
It seems that cbor it trying to fit entire FS_MGMT_DL_CHUNK_SIZE (default 512 bytes) into MCUMGR_BUF_SIZE of buffer which is already partially filled with header data and by default it is 384 bytes in size.

de-nordic, 2020-06-22:
I think this has been fixed her: https://github.com/zephyrproject-rtos/zephyr/pull/23515

nvlsianpu, 2020-09-11:
Thanks @de-nordic!

closed 2020-09-11 by nvlsianpu