zephyrproject-rtos/zephyr #8365: mcumgr: improper response to "image list" command after update.

nvlsianpu, 2018-06-13:
After the update the new image running for test respons with list described only the slot_0 content. After subsequent reset the old image is reverted and the old image responds properly.

Issue reported for smp_svr@nrf52_pca10040

ifillonneau, 2018-06-13:
Hi,
Did you confirm the updated image?
mcumgr --conntype ble --connstring 'peer_name=Zephyr' image confirm

nvlsianpu, 2018-06-25:
Can not reproduce the issue. Last versions I used was https://github.com/runtimeco/mcuboot/commit/5901fd55f5b4d05e7f1b538b454462129a8b1bf1 + https://github.com/zephyrproject-rtos/zephyr/commit/d317a0e6b4b28f1152988f167f2d22f753b307a2 (both master HEADs). This issue was reproduced previously with unknown versions of mcuboot and zephyr.
I'm closing this. Can be reopen if necessary.

closed 2018-06-25 by nvlsianpu