zephyrproject-rtos/zephyr #8374: arm: core: cortex_m: wrongly assumes double precision FPU on Cortex-M7

aurel32, 2018-06-13:
On a Cortex-M7, the FPU support is mandatory, and when it is available it could be simple or double precision. The NXP i.MX RT and Microchip SAM E70 SoC families both have double precision FPU. The STM32F7 family (patches from https://github.com/zephyrproject-rtos/zephyr/pull/7284) have some SoC with single precision FPU and some other with double precision FPU. See http://www.st.com/en/microcontrollers/stm32f7-series.html .

The current build system assumes that all Cortex M7 CPU have a double precision FPU, and passes `-mfpu=fpv5-d16` to GCC in that case (see `cmake/fpu-for-gcc-m-cpu.cmake`). I am fine working on a fix, but I would need some guidance about how to that. Should we use the Kconfig system to define hidden config options like `CPU_HAS_SP_FPU` and `CPU_HAS_DP_FPU`?

galak, 2018-06-14:
Probably adding something in the Kconfig like you suggest that would than get selected by the SoC to convey if it supports double precision or not.  Than using that in cmake to decide for the gcc options is the way to go.

If we can keep it to a single Kconfig option that would be ideal, since on Cortex-M CPU_HAS_FPU/FLOAT imply single precision support.

aurel32, 2018-06-14:
> If we can keep it to a single Kconfig option that would be ideal, since on Cortex-M CPU_HAS_FPU/FLOAT imply single precision support.

I guess you mean on Cortex M4/M33. On Cortex-M7, it imply double precision support.

Looking more in details `CPU_HAS_FPU` is an architecture independent option to unlock the architecture independent floating-point options in `arch/Kconfig`. I am therefore not sure it's a good idea to create a new ARM specific option like `CPU_HAS_FPU_DP`.

Another way of doing that would be to not define `CPU_HAS_FPU` at the board level, but rather select the FPU type, which in turns select CPU_HAS_FPU:

```
config ARMV7_M_ARMV8_M_FPV4_SP
        bool
        # Omit prompt to signify "hidden" option
        depends on CPU_CORTEX_M4
        select ARMV7_M_ARMV8_M_FP
        select CPU_HAS_FPU

config ARMV7_M_ARMV8_M_FPV5_SP
        bool
        # Omit prompt to signify "hidden" option
        depends on CPU_CORTEX_M33 || CPU_CORTEX_M7
        select ARMV7_M_ARMV8_M_FP
        select CPU_HAS_FPU

config ARMV7_M_ARMV8_M_FPV5_DP
        bool
        # Omit prompt to signify "hidden" option
        depends on CPU_CORTEX_M7
        select ARMV7_M_ARMV8_M_FP
        select CPU_HAS_FPU
```

This allows to have a one to one mapping at the cmake level instead of encoding that in `cmake/fpu-for-gcc-m-cpu.cmake`.

galak, 2018-12-13:
@aurel32 any updates on this?

aurel32, 2018-12-13:
I am not very sure about what to do on the Kconfig side. I'll give it another try, and see what people think about it.

galak, 2018-12-13:
@ioannisg do you mind work with @aurel32 on this.

PR #13181 (2019-02-08): Arch arm fix fpu arch selection
Distinguish between single and double precision FPU selection for Cortex-M7 .

Fixes #8374
closed 2019-02-08 by galak