zephyrproject-rtos/zephyr #8379: Support out-of-tree implementations of existing drivers/subsystems

SebastianBoe, 2018-06-14:
It is desired to maintain alternative implementations of existing Zephyr drivers/subsystems out-of-tree.

For this to work the following is needed:

- [ ] Integration with CMake driver build scripts
- [ ] Integration with Kconfig
- [ ] Integration with the userspace/kernelspace separation mechanisms

This is not the same as #5554

SebastianBoe, 2018-06-14:
Driver build scripts often look something like this:

```
zephyr/drivers:
add_subdirectory_if_kconfig(rtc)
```

```
zephyr/drivers/rtc:
zephyr_library()
zephyr_library_sources_ifdef(CONFIG_RTC_QMSI rtc_qmsi.c)
zephyr_library_sources_ifdef(CONFIG_RTC_MCUX rtc_mcux.c)
zephyr_library_sources_ifdef(CONFIG_USERSPACE   rtc_handlers.c)
```

Out-of-tree driver implementations fail when done naively because when CONFIG_RTC is set a CMake library without sources (not allowed) is created in zephyr/drivers/rtc.

SebastianBoe, 2018-06-14:
Should we (can we?) modify zephyr__drivers__rtc from the application to contain the driver? Then we would not have an empty library ...

SebastianBoe, 2018-06-14:
About Kconfig integration. Kconfig driver configuration might look like this today:

> menuconfig RTC
> 	bool "Real-Time Clock"
> 	default n
> 	help
> 	  Enable options for Real-Time Clock drivers.
> 
> if RTC
> 
> config RTC_0_NAME
> 	string "Driver instance name"
> 	default "RTC_0"
> 	depends on RTC && !HAS_DTS
> 	help
> 	  RTC driver instance name
> 
> comment "RTC drivers"
> 
> source "drivers/rtc/Kconfig.qmsi"
> source "drivers/rtc/Kconfig.mcux_rtc"
> 
> endif # RTC
> 

Presumably, we would want to allow an out-of-tree driver to also be configured from the in-tree Kconfig menu for the RTC driver. Intuitively, the user that configures the system does not care if the driver he is using is located in-tree or out-of-tree, and wants to find the configuration options in the same location as in-tree drivers.

This means we need to find some way to inject an out-of-tree Kconfig into the "menuconfig RTC".

SebastianBoe, 2018-06-14:
About userspace/kernelspace separation.

Currently separating symbols into kernelspace and userspace is done by the build system. It uses some simple heuristics to determine if code should be a part of userspace or kernelspace. Described in $ZEPHYR_BASE/CMakeLists.txt:

```
  # Objects default to being in kernel space, and then we exclude
  # certain items.
  set(kernel_object_file_list
    ${ZEPHYR_LIBS_PROPERTY}
    kernel
    )
  list(
    REMOVE_ITEM
    kernel_object_file_list
    app
    )

  # The zephyr libraries in zephyr/lib/ and zephyr/test/ belong in
  # userspace.

  # NB: The business logic for determing what source files are in
  # kernel space and what source files are in user space is
  # fragile. Fix ASAP.
  #
  # The intended design is that certain directories are designated as
  # containing userspace code and others for kernel space code. The
  # implementation we have however is not working on directories of
  # code, it is working on zephyr libraries. It is exploiting the fact
  # that zephyr libraries follow a naming scheme as described in
  # extensions.cmake:zephyr_library_get_current_dir_lib_name
  #
  # But code from test/ and lib/ that is placed in the "zephyr"
  # library (with zephyr_sources()) will not be in a library that is
  # prefixed with lib__ or test__ and will end up in the wrong address
  # space.
  set(application_space_dirs
    lib
    tests
    )
  foreach(f ${kernel_object_file_list})
    foreach(app_dir ${application_space_dirs})
      if(${f} MATCHES "^${app_dir}__") # Begins with ${app_dir}__, e.g. lib__libc
        list(
          REMOVE_ITEM
          kernel_object_file_list
          ${f}
          )
      endif()
    endforeach()
  endforeach()

```

Although technically possible, we cannot naively add an out-of-tree driver implementation to the 'app' library because the 'app' is a part of userspace, but a driver should be in kernelspace.

andrewboie, 2018-06-14:
> Although technically possible, we cannot naively add an out-of-tree driver implementation to the 'app' library because the 'app' is a part of userspace, but a driver should be in kernelspace.

Not necessarily.

What Adithya implemented is that:
- Code in a known set of folders is assumed to always run in supervisor mode and some optimizations can be made
- Otherwise, whether the code runs in supervisor or user mode is not is not known at built time and there is some runtime checking done when system calls are made.

This shouldn't prevent drivers from being built outside of the tree, as long as it's all one build.

SebastianBoe, 2018-06-20:
> Maybe not the right place to ask/suggest this: but why not separating builds for kernel- and user-space if it gets too complicated to put everything under one hood?

Because invoking multiple build systems has a host of problems that should be avoided for as long as is technically possible. See the paper "recursive make considered harmful".

SebastianBoe, 2018-06-20:
> Otherwise, whether the code runs in supervisor or user mode is not is not known at built time and there is some runtime checking done when system calls are made
-- @andrewboie 

Aha, so if a driver is in the 'app' library it will still be in supervisor mode, it'll just have some runtime overhead that won't be present in an in-source driver?

What about memory protection of the kernelspace?

E.g. the linker uses a list of kernel libraries to determine whether something should be placed in kernelspace or userspace. The 'app' is not in this list and will be placed in the APP_DATA_SECTION.

Will this not prevent memory protection of the out-of-source driver?

pboettch, 2018-06-20:
> See the paper "recursive make considered harmful".

See the paper '“Considered Harmful” Essays Considered Harmful'.

But I acknowledge your point, if the chosen strategy of zephyr is to absolutely avoid cmake's `ExternalProject()` and `toolchain-files` (see mailing list), but still stick with cmake, there is not much choice other than to add custom macros and variables and functionalities.

andrewboie, 2018-06-26:
> Aha, so if a driver is in the 'app' library it will still be in supervisor mode, it'll just have some runtime overhead that won't be present in an in-source driver?

Correct. It's just a check to see what mode the CPU is in when a syscall is invoked.

> What about memory protection of the kernelspace?

> E.g. the linker uses a list of kernel libraries to determine whether something should be placed in kernelspace or userspace. The 'app' is not in this list and will be placed in the APP_DATA_SECTION.

Only for the CONFIG_APPLICATION_MEMORY feature.
For global data in the app code that must be in kernel memory, the declaration can be tagged with __kernel and it will be put in the right place.
CONFIG_APPLICATION_MEMORY is going to be deprecated soon.

andrewboie, 2018-06-26:
> Maybe not the right place to ask/suggest this: but why not separating builds for kernel- and user-space if it gets too complicated to put everything under one hood?

This is actually planned at some point soon, so that we can load one or more application binaries at runtime. It will appear around the time we start to implement virtual memory.

However we are over-thinking the current complexities, CONFIG_APPLICATION_MEMORY is going away eventually and nobody doing drivers in their app will be using it.

SebastianBoe, 2018-06-27:
It is good to hear that APPLICATION_MEMORY is going away as it is causing headaches in https://github.com/zephyrproject-rtos/zephyr/pull/8451#discussion_r198137222 .

It might be that to resolve #8451 we would need to rework APPLICATION_MEMORY. Could you elaborate on the plans for it's future so that these efforts are aligned?

Or even better, allow us to drop the feature ...

SebastianBoe, 2018-06-27:
> This is actually planned at some point soon, so that we can load one or more application binaries at runtime. It will appear around the time we start to implement virtual memory.

Note that it is not necessary to invoke CMake / ninja multiple times to achieve this (although it is certainly the easiest approach), there are no CMake-technical blockers for creating multiple application binaries in a single build (but there are many Zephyr-technical blockers that would need to be resolved).

tejlmand, 2018-09-12:
Isn't this related to #8851  ?

carlescufi, 2019-02-04:
@SebastianBoe Does this cover the following use cases:

1. Existing API (`adc.h`) and having an out-of-tree replacement for an existing implementation (say `nrf52`).
2. Existing API (`adc.h`) and having an out-of-tree implementation for a SoC that is supported by Zephyr but there's no driver for it upstream
3. Existing API (`adc.h`) and having an out-of-tree implementation for a SoC that is **not** supported by Zephyr and obviously there's no driver for it upstream
4. New driver class/API completely out-of-tree

SebastianBoe, 2019-02-11:
The three first are covered by this issue. And now that APPLICATION_MEMORY is gone, I suspect this might be a simple matter of dropping the "empty library" check. But I would need to prototype to make sure.

The last use-case is not covered by this issue. It is covered by https://github.com/zephyrproject-rtos/zephyr/issues/5554

carlescufi, 2019-02-11:
@SebastianBoe is the last use-case really covered by #5554? I mean, does it take into account the necessary additions to the DT bindings and in general the DT integration?

SebastianBoe, 2019-02-11:
I can't speak about how to do out-of-tree bindings, I don't believe it is supported.

I believe #5554 was assuming the driver would not use anything from DT, which may or may not have been reasonable a year ago.

carlescufi, 2019-03-14:
@SebastianBoe can you update this?

SebastianBoe, 2019-03-14:
> https://github.com/zephyrproject-rtos/zephyr/issues/8379#issuecomment-397255527

This proposal has been implemented and fixes the CMake integration of out-of-tree drivers.

SebastianBoe, 2019-03-14:
> https://github.com/zephyrproject-rtos/zephyr/issues/8379#issuecomment-397256844

Injecting out-of-tree Kconfig into in-tree Kconfig menu's has not been implemented. But technically, you don't need this to have an out-of-tree driver if one accepts that the menu options are placed out-of-tree.

Cleaner menu integration can/should be a separate issue.

So the Kconfig integration is also OK.

SebastianBoe, 2019-03-14:
Closing as resolved.

Further improvements can be reported as separate issues.

closed 2019-03-14 by SebastianBoe

LuoZhongYao, 2019-11-22:
Is there a document on how to use the out-of-tree driver.

SebastianBoe, 2019-11-22:
Not really.

To do this you need to use two things, the zephyr_library API, and the ZEPHYR_EXTRA_MODULES feature.

The zephyr_library API I assume is familiar. ZEPHYR_EXTRA_MODULES is documented here:

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/guides/modules.html

LuoZhongYao, 2019-11-23:
The tree-out driver can be compiled with ZEPHYR_EXTRA_MODULES, but Kconfig is a separate menu that can put Kconfig outside the tree in the same position as the tree.
E.g:
The GPIO driver Kconfig is in app/drivers/gpio/Kconfig.foo, the option is in menuconfig -> Device Drivers -> FOO GPIO Driver
The out-of-tree driver has the same structure as the tree, and it is more convenient to merge the tree drive into the tree.