zephyrproject-rtos/zephyr #8393: `CONFIG_MULTITHREADING=n` builds call `main()` with interrupts locked

carlescufi, 2018-06-14:
For consistency with multithreaded builds, `main()` should be called with interrupts unlocked and ready to go.

andyross, 2018-06-14:
The root cause here is that where the default Zephyr thread startup sets the lock state to a known value inside arch code, where we are at the end of _Cstart() is unspecified.  We never took a lock, so we can't release it.

There needs to be an arch hook for "enable interrupts unconditionally", either a new API call or a special key value that can be passed to irq_unlock().

carlescufi, 2018-06-15:
Raising priority because this cannot be easily worked around on the application code in a way that is cross-arch and clean.

carlescufi, 2018-06-15:
@agross-linaro @andyross I also notice that even when enabling interrupts I see behavior that is not explained by my code. Could it be that the ISR wrapper in https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/arm/core/isr_wrapper.S#L42 also needs conditional compilation based on `CONFIG_MULTITHREADING`? 
Right now this is completely preventing proper use of MCUboot.

agross-linaro, 2018-06-15:
If no multithreading then yeah this shouldn't be setting the context switch.

carlescufi, 2018-06-15:
@agross-linaro I've disabled `CONFIG_SYS_POWER_MANAGEMENT` and `CONFIG_PREEMPT_ENABLE` and I believe that should compile out all of the code we don't need, though confirmation would be appreciated.

carlescufi, 2018-06-18:
Added an MCUboot issue to be resolved once this one is closed: https://github.com/runtimeco/mcuboot/issues/302

andyross, 2018-06-28:
Drop priority as there's an active workaround and it's not blocking work.  It remains a real bug and needs a fix for 1.13.

nashif, 2018-08-26:
@andyross ping

carlescufi, 2019-05-09:
Reopening this since the fix was reverted in master

andyross, 2019-08-20:
Coming by cleaning up my bug list: what's the desired fix here?  I forget the details of @nashif's reversion.  The interrupt fix alone seemed relatively benign, though I know there's an argument about whether or not to support MULTITHREADING=n as an official target.

nordic-krch, 2020-07-10:
any updates there? Just hit this issue again.

pabigot, 2020-07-10:
I've updated this to medium based on https://github.com/zephyrproject-rtos/zephyr/issues/8393#issuecomment-525359687 which appears to be immediately followed by changing it to low instead.  Personally, I'd call it high so it doesn't get overlooked as it has for the last two years.  (Apparently disabling threading is pretty unusual, but it should still work.)

pabigot, 2020-07-21:
More context: we need the Nordic QSPI driver, which uses interrupts, to notify of completion of an operation.  We don't need anything else that comes with multithreading because we can busy-wait for the notification.

pabigot, 2020-07-22:
Some historical context: this was "solved" in #9668, reverted in #9829, and re-introduced via #9808 as a proposal to remove this feature.

Perhaps further discussion should be in #9808.  An example of what I want to be able to do us to make `samples/basic/blinky` work as expected with the following changes:
```
diff --git a/samples/basic/button/prj.conf b/samples/basic/button/prj.conf
index 91c3c15b37..497efa4a8c 100644
--- a/samples/basic/button/prj.conf
+++ b/samples/basic/button/prj.conf
@@ -1 +1,2 @@
 CONFIG_GPIO=y
+CONFIG_MULTITHREADING=n
diff --git a/samples/basic/button/src/main.c b/samples/basic/button/src/main.c
index 1e9f9579e1..fba2b955a1 100644
--- a/samples/basic/button/src/main.c
+++ b/samples/basic/button/src/main.c
@@ -12,7 +12,7 @@
 #include <sys/printk.h>
 #include <inttypes.h>
 
-#define SLEEP_TIME_MS  1
+#define SLEEP_TIME_MS  1000
 
 /*
  * Devicetree helper macro which gets the 'flags' cell from a 'gpios'
@@ -90,10 +90,10 @@ void main(void)
 
        led = initialize_led();
 
-       printk("Press the button\n");
        while (1) {
+               printk("Press the button\n");
                match_led_to_button(button, led);
-               k_msleep(SLEEP_TIME_MS);
+               k_busy_wait(USEC_PER_SEC / MSEC_PER_SEC * SLEEP_TIME_MS);
        }
 }
```

nashif, 2020-07-23:
We need to step back and think why we wanted or want `CONFIG_MULTITHREADING=n` in the first place. What is the requirement here and what is the usecase?
If I am not mistaken, the only reason why we want this is footprint and being able to fit the bootloader built with Zephyr in limited flash available for bootloader.

So back to what I say in #9808: 
> Reduce complexity and provide just multi-threading support and focus on reducing footprint for all users.

if we decide to keep `CONFIG_MULTITHREADING=n`, this needs to come with a clear definition and list of constraint. We should not in any case go beyond those constraints as needs arise, if we start enabling more and more features when scheduler is disabled, we will end up with the original problem of footprint.

Maybe a definition like this:
- No scheduling, single thread support
- Interrupt handling? What are the constraints?
- No synchronization, obviously
- What about sleeping? Define constraints

pabigot, 2020-07-23:
For the purpose of discussion here's the size difference when adding `CONFIG_MULTITHREADING=y` to mcuboot on `particle_xenon` (nRF52840, 4 KiBy pages).  It's a 21% increase of 7764 bytes flash.  The minimum partition for mcuboot must increase from 9 pages to 11 pages (12 is the default).

```
CONFIG_MULTITHREADING=n
Memory region         Used Size  Region Size  %age Used
           FLASH:       36760 B        48 KB     74.79%
            SRAM:       22536 B       256 KB      8.60%
        IDT_LIST:          72 B         2 KB      3.52%

CONFIG_MULTITHREADING=y
Memory region         Used Size  Region Size  %age Used
           FLASH:       44524 B        48 KB     90.58%
            SRAM:       25240 B       256 KB      9.63%
        IDT_LIST:          72 B         2 KB      3.52%
```

On `nrf52840dk_nrf52840` (pca10056) which includes the QSPI driver that requires delivered interrupts to function it's a 22% increase of 8452 bytes flash.  The minimum partition for mcuboot must increase from 10 pages to 12 pages (12 is the default).

```
CONFIG_MULTITHREADING=n
Memory region         Used Size  Region Size  %age Used
           FLASH:       38984 B        48 KB     79.31%
            SRAM:       23832 B       256 KB      9.09%
        IDT_LIST:          88 B         2 KB      4.30%

CONFIG_MULTITHREADING=y
Memory region         Used Size  Region Size  %age Used
           FLASH:       47436 B        48 KB     96.51%
            SRAM:       26552 B       256 KB     10.13%
        IDT_LIST:          88 B         2 KB      4.30%
```

nashif, 2020-07-23:
please provide the complete config for each.

pabigot, 2020-07-23:
> please provide the complete config for each.

That would be pointless and messy.  Add `CONFIG_MULTITHREADING=y` to the bottom of mcuboot's `boot/zephyr/prj.conf` and see what it does on platform and version you're concerned with.

pabigot, 2020-07-23:
> An example of what I want to be able to do us to make `samples/basic/blinky` work as expected
with `CONFIG_MULTITHREADING=n`.

I've tried updating 17e9d623b417d7a0e9b32ca9b5c4b6cd89062ec0 to apply to current master; the button interrupt is delivered but the application does not continue in the main loop: `z_arm_usage_fault()` occurs.  Same when just using `irq_unlock(0)` in main.

mcuboot has been patched to enable interrupts conditionally in the patch below.  The behavior I see with GPIO interrupts makes me concerned this is more fragile than we believe.  It may well be time to give up on `CONFIG_MULTITHREADING=n` in Zephyr, and tell people who want small footprints to use a different solution.

```
 commit 5e48c55df9bec29a0d44db4d5da65ca4c24a011a
Author: Carles Cufi <carles.cufi@nordicsemi.no>
Date:   Fri Jun 15 12:58:08 2018 +0200

    zephyr: Enable interrupts unconditionally
    
    Due to an issue described here:
    https://github.com/zephyrproject-rtos/zephyr/issues/8393
    interrupts are not enabled when multithreading is disabled.
    Enable interrupts to allow the serial recovery mode UART to receive
    characters.
    
    Note: This commit must be reverted once the issue is addressed.
    
    Signed-off-by: Carles Cufi <carles.cufi@nordicsemi.no>

diff --git a/boot/zephyr/serial_adapter.c b/boot/zephyr/serial_adapter.c
index e04e0b1..f51dc94 100644
--- a/boot/zephyr/serial_adapter.c
+++ b/boot/zephyr/serial_adapter.c
@@ -197,5 +197,10 @@ boot_uart_fifo_init(void)
 
        uart_irq_rx_enable(uart_dev);
 
+       /* Enable all interrupts unconditionally. Note that this is due
+        * to Zephyr issue #8393. This should be removed once the
+        * issue is fixed in upstream Zephyr. */
+       irq_unlock(0);
+
        return 0;
 }
```

nashif, 2020-07-23:
> That would be pointless and messy. Add `CONFIG_MULTITHREADING=y` to the bottom of mcuboot's `boot/zephyr/prj.conf` and see what it does on platform and version you're concerned with.

why is this pointless? It shows that you for example are measuring with CONFIG_DEBUG enabled, so for the above board I get for:
``` west build -b particle_xenon -- -DCONFIG_MULTITHREADING=y -DCONFIG_DEBUG=n```

```
[232/237] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       38124 B        48 KB     77.56%
            SRAM:       25240 B       256 KB      9.63%
        IDT_LIST:          72 B         2 KB      3.52%
```

looking at ram_report...

kernel is 4290 (11.41%)
logging is 3926 (10.44%)
drivers is 4862 (12.93%)
mbedtls is 9738 (25.90%)

etc.

so kernel and scheduler are not the major portion here and I am sure we can save more by applying some kconfigs to reduce kernel/scheduler footprint, for example...

```
CONFIG_TIMESLICING=n
CONFIG_SIMPLE_FATAL_ERROR_HANDLER=y
CONFIG_NUM_PREEMPT_PRIORITIES=0
CONFIG_NUM_COOP_PRIORITIES=1
```

nashif, 2020-07-23:
goes doen even further if we disable logging and instead use CONFIG_LOG_MINIMAL=y

```
[228/234] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       32304 B        48 KB     65.72%
            SRAM:       23856 B       256 KB      9.10%
        IDT_LIST:          72 B         2 KB      3.52%
```

does not build out of the box, there some loggin code in mcuboot setting up threads and other things they will not link, not sure why all of this is needed if we are running in a single thread.

pabigot, 2020-07-23:
It would have been pointless to provide the whole configuration here; a little more detail on what was done covered it.  But your observation that the default configuration is not optimized is well-taken.

From a baseline of the best overrides that produce a linked image:
```
CONFIG_DEBUG=n
CONFIG_TIMESLICING=n
CONFIG_NUM_PREEMPT_PRIORITIES=0
CONFIG_NUM_COOP_PRIORITIES=1
```

the incremental cost of `CONFIG_MULTITHREADING=y` is:
```

Memory region         Used Size  Region Size  %age Used
           FLASH:       31776 B        48 KB     64.65%
            SRAM:       22536 B       256 KB      8.60%
        IDT_LIST:          72 B         2 KB      3.52%
```
to
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       37612 B        48 KB     76.52%
            SRAM:       25224 B       256 KB      9.62%
        IDT_LIST:          72 B         2 KB      3.52%
```
which is 18.4% at 5836 bytes.

The relative cost seems pretty constant at roughly 20%.  I'm not going to play with more combinations because as I noted on #9808 I support removing the option to disable multithreading.

nashif, 2020-07-23:
> which is 18.4% at 5836 bytes.

Hmm, I measure 4290 bytes for the whole kernel above, so most of this overhead is probably coming from somewhere else and when multithreading is enabled, probably thru logging. With log minimal enable I get;

```
[109/114] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       33780 B        48 KB     68.73%
            SRAM:       25120 B       256 KB      9.58%
        IDT_LIST:          88 B         2 KB      4.30%
```

nashif, 2020-07-23:
I think in next dev-review we should start a session of puncover on a common configuration (mcuboot or whatever) and go line by line and identify where we can save bytes... :) This will be valuable for everyone, not only for mcuboot IMO

carlescufi, 2020-07-24:
@nashif I am very much in favor of correctly defining what disabling multithreading means, and implementing it properly as per the set of requirements that we decide to include.
 
> if we decide to keep `CONFIG_MULTITHREADING=n`, this needs to come with a clear definition and list of constraint. We should not in any case go beyond those constraints as needs arise, if we start [...]

Here is my take on what this would be if we made it happen:

When CONFIG_MULTITHREADING is disabled, the following functionality is provided:

- Booting the system to a `main()` function, including the initialization of the different memory segments/areas
- Being able to define interrupt handlers in the same way that multithreaded applications do
- Being able to use the interrupt-handling APIs in Zephyr (`irq_lock()`, `irq_unlock()`, etc)
- Being able to schedule pre-`main()` executable functions (`SYS_INIT`) that make no use of the multithreaded functionality 

The following will *not* be supported:
- Any sort of interaction with drivers that do not take into account `CONFIG_MULTITHREADING`
- Kernel timers and timeouts? (TBD, but I don't think this is feasible)
- Kernel threads, synchronization and message-passing primitives
- Any Zephyr subsystems and libraries that make use of the above (most of them)

nashif, 2020-07-24:
@carlescufi  would be great to specify why all of this is needed, are we still going after the few kilobytes from the kernel or is there some other rationale for having this supported?

carlescufi, 2020-07-24:
> @carlescufi would be great to specify why all of this is needed, are we still going after the few kilobytes from the kernel or is there some other rationale for having this supported?

The idea is to be able to use Zephyr as an almost "bare-metal" development platform. Threads and threading can be expensive in RAM and ROM footprint, but Zephyr provides a lot of value beyond its RTOS underpinnings:

- A flexible build and configuration system
- A hardware description language along with an API to access it
- Platform and architecture-agnostic primitives for basic operations such as booting and handling interrupts

I personally believe that maintaining the possibility of using Zephyr in this way it's added value at a (relative) low cost for the project, so I would be very much in favor of formally defining the requirements and limitations and then officially supporting this "feature".

carlescufi, 2020-07-24:
> Hmm, I measure 4290 bytes for the whole kernel above, so most of this overhead is probably coming from somewhere else and when multithreading is enabled, probably thru logging. With log minimal [...]

There are some additional benchmarks [here](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/basic/minimal#building-and-measuring-rom-size).

nashif, 2020-09-03:
single threading is scheduled for deprecation, so lower priority.

pabigot, 2020-09-03:
Why keep this open?  It's unlikely that the decision will change, and if it does part of the task will be to identify all the things that need to work, in which case this issue will be rediscovered.  Keeping it open affects the low bug count which can now impact release.

I'd rather close all these issues and move on.  Maybe discuss next triage.

nashif, 2020-09-23:
clearing assignee as this feature is being deprecated, we can talk about this next and discuss if someone will step up and own this and if those issues will be assigned to them. The bug is open, but the code is still in the tree, even with deprecation. Deprecated code does not mean we stop fixing bugs.

pabigot, 2020-09-23:
@ioannisg re our discussion earlier today.

carlescufi, 2021-04-08:
reopening as this still applies

ioannisg, 2021-05-19:
@carlescufi you reopened this, however, I am not sure if this still applies. CONFIG_MULTITHREADING=n is only maintained on Cortex-M, for now, and I can confirm that in Cortex-M this is not an issue.

carlescufi, 2021-05-19:
> @carlescufi you reopened this, however, I am not sure if this still applies. CONFIG_MULTITHREADING=n is only maintained on Cortex-M, for now, and I can confirm that in Cortex-M this is not an issue.

I agree, as long as you confirm that this is not an issue with Cortex-M then we can close this @ioannisg

ioannisg, 2021-05-19:
https://github.com/zephyrproject-rtos/zephyr/commit/3b89cf173b6b0085c84a09fd97b42e08a25ae4b3

@carlescufi ^^

carlescufi, 2021-05-21:
Thanks for confirming @ioannisg. Closing this now.

PR #9668 (2018-08-27): kernel: Enable interrupts for MULTITHREADING=n on supported architect…
…ures

Some applications have a use case for a tiny MULTITHREADING=n build
(which lacks most of the kernel) but still want special-purpose
drivers in that mode that might need to handle interupts.  This
creates a chicken and egg problem, as arch code (for obvious reasons)
runs _Cstart() with interrupts disabled, and enables them only on
switching into a newly created thread context.  Zephyr does not have a
"turn interrupts on now, please" API at the architecture level.

So this creates one as an arch-specific wrapper around [... cut]
commit 17e9d623 (2018-08-27): kernel: Enable interrupts for MULTITHREADING=n on supported arch's

Some applications have a use case for a tiny MULTITHREADING=n build
(which lacks most of the kernel) but still want special-purpose
drivers in that mode that might need to handle interupts.  This
creates a chicken and egg problem, as arch code (for obvious reasons)
runs _Cstart() with interrupts disabled, and enables them only on
switching into a newly created thread context.  Zephyr does not have a
"turn interrupts on now, please" API at the architecture level.

So this creates one as an arch-specific wrapper around
_arch_irq [... cut]
closed 2021-05-21 by carlescufi