zephyrproject-rtos/zephyr #8409: Pin interrupt not handled when two pin ints fires in quick succession

alexsven, 2018-06-15:
I'm seeing an issue where one pin interrupt is not handled when there is a second pin interrupt firing within 12µs after the first. I have attached a logic_analyzer file that shows the two pin interrupts and the handler of the first pin interrupt. 
[Interrupt_crash.zip](https://github.com/zephyrproject-rtos/zephyr/files/2105881/Interrupt_crash.zip)

Explanation of the logic signals: The pin interrupts are active LOW. I've added a pin toggling to the handler that sets the pin LOW when the handler is called (will send a k_alert_send), the pin toggles back to HIGH when k_alert_recv for the same signal is called.

pizi-nordic, 2018-06-15:
Could you please attach your project configuration?

pizi-nordic, 2018-06-15:
@jarz-nordic: Do you know who could look at this?

jarz-nordic, 2018-06-18:
@kl-cruz : may you please take a look?

pizi-nordic, 2018-06-29:
@kl-cruz: Any progress?

kl-cruz, 2018-07-02:
I missed it in notifications! Please give me a while

kl-cruz, 2018-07-26:
I need some sample code. I can't reproduce it...

koffes, 2018-07-27:
Apologies, but we have had to prioritise other issues lately. I have created a ticket for this in our internal Jira as well, and I will argue we prioritise giving you example code for our next sprint. I can at least confirm that we are still seeing this issue on a 52832.

alexsven, 2018-08-29:
I have been trying to recreate this problem with a set of DKs, connecting two GPIOs, so that a pin toggle on one side would be seen as an interrupt on the other DK. By inserting some code in between toggling pin 15 and pin 16 I am able to change the timing between them. I can see, by using a logic analyzer that it cycles between 0.88µs and 19.68µs, but I can't reproduce the bug.

In the actual code there are a lot more going on, threads being scheduled, SPI traffic running etc, so it might be that this example is way to simple and that is why we can't reproduce it. It's not a big concern for us anymore as we have implemented watchdogs to fix the behaviour, but I've included the code I used for testing, if that can help you in the right direction.
[interrupt_bug.zip](https://github.com/zephyrproject-rtos/zephyr/files/2331517/interrupt_bug.zip)

anangl, 2018-09-07:
@nashif I took a look. I even spent quite a lot of time on this issue trying to replicate the bug but I didn't succeed. I tried with a similar setup as @alexsven, in a loop with increasing interval from ~0.64 µs to ~20.42 µs with the step of one cycle of the 64 MHz clock, i.e. 15.625 ns. And both callbacks were always called. In certain circumstances the order in which they were called was  different (as events from both pins generate the same interrupt and sometimes they can be noticed during one interrupt handler execution), but I haven't noticed a missing callback. So, when it comes to gpio callbacks execution, everything seems to be fine.
Maybe the issue is related to something else in this more complicated setup where it has been seen?

anangl, 2018-09-11:
Closing this as currently no one can replicate such behavior with some sample code.
It can be re-opened once such code snippet appears.

closed 2018-09-11 by anangl