zephyrproject-rtos/zephyr #8419: Bluetooth: tester: LPN Poll issue

MariuszSkamra, 2018-06-15:
There are 4 test cases that fail because PTS is not receiving LPN Poll requests.
Affected test cases:
MESH/NODE/FRND/LPN/BI-02-C, MESH/NODE/FRND/LPN/BV-02-C, MESH/NODE/FRND/LPN/BV-03-C, MESH/NODE/FRND/LPN/BV-06-C
Tested on  nrf52840_pca10056
One need to try run those tests manually using mesh_shell application to see if the fail is caused by incorrect tester application config.

[MESH/NODE/FRND/LPN/BV-02-C](https://github.com/zephyrproject-rtos/zephyr/files/2106559/RTT_Terminal_201806151720.log)

nashif, 2018-09-03:
@MariuszSkamra will this be fixed for 1.13? @jhedberg fyi

galak, 2018-12-13:
@MariuszSkamra is this still showing up with recent runs of PTS.

@jhedberg should be addressed with Vendor Extension Support.

laperie, 2019-03-14:
We'll likely to resolve this issue after Vendor Extensions are done; moving to past 1.14

carlescufi, 2019-03-19:
Adding @joerchan to keep him in the loop when it comes to Mesh extensions.

carlescufi, 2019-03-19:
@jhedberg there are 3 commands implemented in the legacy LL already, you could start using those in the Host.

jhedberg, 2019-03-19:
> @jhedberg The mesh_advertise and mesh_advertise_cancel commands are implemented in the legacy LL already, you could start using those in the Host.

@carlescufi yes, I could take a look if we can replace the bt_le_adv_start/stop usage with that (at least for a combined build where we know the controller supports the vendor extensions). However, I'd feel uneasy making this never-before used controller functionality the primary method of mesh transmission just a few weeks before LTS, so I think I'd prefer to experiment with this only once we have the release out.

carlescufi, 2019-08-15:
@MariuszSkamra is this still an issue?

MariuszSkamra, 2019-08-16:
No, this is not valid

closed 2019-08-16 by MariuszSkamra