zephyrproject-rtos/zephyr #8438: cmake: Propagation of library specific compile flag

tejlmand, 2018-06-18:
When enabling:                                                                                      
(top menu) → Bluetooth → Advanced features                                                          
[*] Fast Encryption Setup                                                                           
                                                                                                    
I noticed that the -Ofast is propagated to parts of the build system which has a dependency to the  
library subsys__bluetooth__controller.                                                                 
                                                                                                       
In this particular case, this happens for zephyr_prebuilt.elf and zephyr.elf, where the files          
empty_file.c and isr_tables.c is also compiled with -Ofast.                                   

This is not critical in this particular build, but I believe it is unintended that those files 
were compiled with -Ofast.                                                                
                                                                                                       
The corresponding flag for this setting in KConfig and CMake is: CONFIG_BT_CTLR_FAST_ENC                                                       
                                                                                                       
Output from Ninja, where it is visible that the two files mentioned are compiled with -Ofast.          
 [146/148] ccache /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc -DBUILD_VERSION=v1.12.0-191-g49554dd -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../../kernel/include -I../../../../arch/arm/include -I../../../../arch/arm/soc/nordic_nrf/nrf52 -I../    ../../../arch/arm/soc/nordic_nrf/nrf52/include -I../../../../arch/arm/soc/nordic_nrf/include -I../../../../boards/arm/nrf52840_pca10056 -I../../../../include -I../../../../include/drivers -Izephyr/include/generated -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi    /6.2.0/include -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include-fixed -I../../../../lib/libc/minimal/include -I../../../../ext/lib/crypto/tinycrypt/include -I../../../../ext/hal/cmsis/Include -I../../../../ext/hal/nordic/nrfx -I../../../../ext/hal/n    ordic/nrfx/drivers/include -I../../../../ext/hal/nordic/nrfx/hal -I../../../../ext/hal/nordic/nrfx/mdk -I../../../../ext/hal/nordic/. -I../../../../subsys/bluetooth -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /projects/github/zephyr/samples/bluetooth/peripheral_hr/build/zephyr/includ    e/generated/autoconf.h -ffreestanding -Wno-main --sysroot /opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr -mthumb -mcpu=cortex-m4 -std=c99 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-in    t -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -Ofast -MD -MT zephyr/CMakeFiles/kernel_elf.dir/misc/empty_file.c.obj -MF zephyr/CMakeFiles/kernel_elf.dir/misc/empty_file.c.obj.d -o zephyr/CMakeFiles/kernel_elf.dir/misc/empty_file.c.obj   -c /projects/github/zephyr/misc/empty_file.c
 166 [147/148] ccache /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc -DBUILD_VERSION=v1.12.0-191-g49554dd -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../../kernel/include -I../../../../arch/arm/include -I../../../../arch/arm/soc/nordic_nrf/nrf52 -I../    ../../../arch/arm/soc/nordic_nrf/nrf52/include -I../../../../arch/arm/soc/nordic_nrf/include -I../../../../boards/arm/nrf52840_pca10056 -I../../../../include -I../../../../include/drivers -Izephyr/include/generated -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi    /6.2.0/include -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include-fixed -I../../../../lib/libc/minimal/include -I../../../../ext/lib/crypto/tinycrypt/include -I../../../../ext/hal/cmsis/Include -I../../../../ext/hal/nordic/nrfx -I../../../../ext/hal/n    ordic/nrfx/drivers/include -I../../../../ext/hal/nordic/nrfx/hal -I../../../../ext/hal/nordic/nrfx/mdk -I../../../../ext/hal/nordic/. -I../../../../subsys/bluetooth -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /projects/github/zephyr/samples/bluetooth/peripheral_hr/build/zephyr/includ    e/generated/autoconf.h -ffreestanding -Wno-main --sysroot /opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr -mthumb -mcpu=cortex-m4 -std=c99 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-in    t -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -Ofast -MD -MT zephyr/CMakeFiles/kernel_elf.dir/isr_tables.c.obj -MF zephyr/CMakeFiles/kernel_elf.dir/isr_tables.c.obj.d -o zephyr/CMakeFiles/kernel_elf.dir/isr_tables.c.obj   -c zephyr/isr_tables.c

SebastianBoe, 2018-06-20:
Thank you for reporting.

This is indeed a bug.

flags added through zephyr_library_compile_options should only be added to the library, not to the interface of the library.

I have a fix and an idea of the root cause.

The elf target inherits the interface properties of the subsys__bluetooth__controller. This is expected and desired behaviour.

The subsys__bluetooth__controller library links with a special library that only contains the Ofast property to obtain it. This is also expected and desired behaviour.

But, when the subsys__bluetooth__controller library links with the special library it should do it in such a way that it's interface is not changed. Only it's private properties. This is what target_link_libraries(<target> PRIVATE <item>) does. But we have been using target_link_libraries(<target> <item>), which is equivalent to target_link_libraries(<target> PRIVATE <item>) AFAICT.

Changing to use private linking as seen in the patch below resolves the issue, but creates some other issues that I need to figure out before we can resolve this ...

https://github.com/SebastianBoe/zephyr/commit/e89edc989ddd3439d573f928e758ca5aa984afe4

tejlmand, 2018-06-21:
Hi,

Please note that RFC #8451 solve this problem.
https://github.com/tejlmand/zephyr/blob/82c53caf79d693a2f37b0752702cb11af78b0615/subsys/bluetooth/controller/bluetooth_controller.cmake#L49-L55
as flags are set directly on the source files (which has precedence over the other flags)

In CMake you have
CMAKE_<LANG>_FLAGS, e.g. CMAKE_C_FLAGS
target_compile_options, on libs or executables
Source file properties, by using SET_SOURCE_FILES_PROPERTIES(<src> PROPERTIES COMPILE_FLAGS <flags>)

Source file properties takes precedence over compile options on a lib/exe. (and will not be propagated)
Compile options on a lib/exe takes precedence over CMAKE_C_FLAGS (and can be propagated)

Thus, specifying those compile options on the files ensures:
1) They take precedence over other flags
2) They are not propagated
3)No need to any workarounds using interface libraries and ordering for placing flags on certain files.

SebastianBoe, 2018-06-21:
That's great. #8451 is further down on my TODO-list, will get to it soon.

SebastianBoe, 2018-06-22:
I have a fix now. Just need to wait for 2 PR's to go in first.

#8531 
#8532

Proper fix:
https://github.com/SebastianBoe/zephyr/tree/proper_fix_of_8438

carlescufi, 2018-06-26:
@SebastianBoe the 2 PRs you mention are now merged

SebastianBoe, 2018-06-26:
Ah, I was mistaken, the fix is also blocked by https://github.com/zephyrproject-rtos/zephyr/pull/8533

PR #9409 (2018-08-14): cmake: Don't propagate zephyr_library_compile_options
zephyr_library_compile_options will apply compile options to a library
with high priority. Due to bug #8438 these compile options have been
applied not only to the library, but also to it's public interface and
therefore to any library that has been linking with the library.

With this patch we now use PRIVATE linking instead of PUBLIC linking
this fixes #8438 .
commit 83d2a47d (2018-08-14): cmake: Don't propagate zephyr_library_compile_options

zephyr_library_compile_options will apply compile options to a library
with high priority. Due to bug #8438 these compile options have been
applied not only to the library, but also to it's public interface and
therefore to any library that has been linking with the library.

To resolve #8438 we use PRIVATE linking instead of PUBLIC linking.
closed 2018-08-14 by nashif