zephyrproject-rtos/zephyr #8441: cmake: Improve library dependencies by reducing number of libraries, whole-archive, start/end-group

tejlmand, 2018-06-18:
Zephyr subsystems creates multiple (sub-)sublibraries, especially bluetooth and
net. Those subsystems also has internal dependencies to each other.

Multiple subsystems in Zephyr has a lot of sub libraries, which I believe
should be minimized, especially a KConfig is good for configuring the libraries.

Subsystem Bluetooth contains:
subsys_bluetooth, subsys__bluetooth__common, subsys__bluetooth__controller,
subsys__bluetooth__host__mesh, subsys__bluetooth__host

Subsystem net contains:
subsys__net, subsys__net_ip, subsys__net_ip_l2, subsys__net_ip_l2_ethernet,
subsys__net_ip_l2_ieee802154, subsys__net_ip_l2_openthread,
subsys__net__lib__app, subsys__net__lib__dns, subsys__net__lib__http,
subsys__net__lib__lwm2m, subsys__net__lib__mqtt,  subsys__net__lib__websocket 

When working in an IDE, all those libraries are presented to the user.
Such libraries are often displayed at the top of the tree in an IDE.
A general user of Zephyr must must be expected to primarly work on libapp and 
his own libraries and therefor it is not a good experience to be presented with
so many sub-libraries.
Having a single bluetooth or network library would be much nicer.

Furthermore Zephyr links everything together using -Wl,--start-group 
<libraries> -Wl,--end-group, as well as Zephyr build system uses 
-Wl,--whole-archive.

This principle has several drawbacks;
The -Wl,--whole-archive drawback is also described in: #6961

Creating executables in this manner requires extra build scripts and 
is non-standard for a CMake project. Instead of using the CMake API 
to link together libraries into an executable with
target_link_libraries() one must have build scripts that ensures the 
linker is passed this list of linker flags.

It introduces two kinds of libraries, instead of having just one kind. 
Libraries which must be whole-archived, and libraries that should not.

There is also a theoretical code size performance impact. If 
whole-archive is used and there exists an object file in a library 
that happens to not be in use (dead code) the object file will still 
be included in the executable, instead of being pruned as one would 
want it to be. But if one links without '--whole- archive' only the 
object files that have symbols needed by the link will be used.
In Zephyr, the handling of library content is well handled using 
KConfig, and doesn't seem to suffer from this risk.
However, that doesn't mean the problem may not be present in certain 
parts of Zephyr.

Finally, using a flat linking model where all libraries are included 
means that we do not have readily available information about how 
libraries depend on each other.
CMake list ${ZEPHYR_LIBS_PROPERTY} is furthermore a global variable
to which libraries are added and later used in the linking stage.
Use of global variables should be minimized whenever possible.

SebastianBoe, 2018-06-26:
Would you mind splitting the "many libraries causes noise in IDE's" out into it's own issue?

tejlmand, 2018-07-10:
> Would you mind splitting the "many libraries causes noise in IDE's" out into it's own issue?

Done: see #8826

tejlmand, 2018-08-07:
> Great that 8826 was created. But could we also remove the parts of 8826 that are redundant with this issue?
>
> e.g. remove the text above
>
> "Furthermore Zephyr links everything together using -Wl,--start-group"

Removed the text directly related to the IDE, but kept text which describes/lists the sub-libraries, as that is important for the issue. With many sub-libraries, it becomes hard to setup all inter dependencies (and are they circular as well ?)
Example,
every subsys__bluetooth_X probably depends upon subsys__bluetooth__common
but are there inter dependencies between subsys__bluetooth__controller and subsys__bluetooth__host ?
subsys__bluetooth__host__mesh and subsys__bluetooth__host 

Thus, reducing this to single subsys_bluetooth lib removes the need in CMake to described such relations, as well as it makes a cleaner and easier to describe build system, in terms of what libraries are build and what do they provide.

> and change the title to
>
> "cmake: Improve library dependencies by removing whole-archive, start/end-group"

Done