zephyrproject-rtos/zephyr #8454: Xtensa: State of user defined TIEs is not stored on context switch (swap)

aborany, 2018-06-18:
Xtensa Cores allow to define user specific core extensions (TIEs). Some of them may contain internal states which need to be preserved over context/thread switches. Currently the state of the user TIEs are not stored  properly on the thread stack.

We see that _xt_context_restore function is called during zxt_dispatch, which is trying to restore the user TIE state from the new thread, but before the current state was not saved within the __swap function. Therefore the state is then also not restored properly, when reentering the context/thread again.

Assumption is that xthal_save_extra_nw function should be called within the swap function. Maybe the stack frame structure must be extended to give space for storage.

andyross, 2018-06-19:
What platform are you on?  This bug is against the legacy xtensa layer and not asm2 (which, to be fair, also doesn't support arbitrary SR extensions, but it doesn't use the HAL in these contexts either).

There's no general solution to this problem, really.  We can support any combination of special registers in the stack frame, but we need to know what the requirement is and on what hardware.

aborany, 2018-06-20:
We are using Xtensa architecture with our custom core configuration. I agree that the issue is within the xtensa adaptation layer.

I see that in file arch/xtensa/xtensa_context.s within the function _xt_context_save the according registers would be saved,  but this function is only called when getting an interrupt.
In file arch/xtensa/swap.s in contrast we have a special handling for saving the current context registers, but here the user extension TIE registers are not saved before calling _zxt_dispatch, which the already calls _xt_context_restore. My assumption is that xthal_save_extra_nw should be called in swap.s as well or the context_save should be called there, but I am not sure about some optimization done in swap.s to save runtime by storing not all registers like during interrupt.

andyross, 2019-02-26:
Nitpicky process point, as we scrub issues for 1.14:  this is really a feature request, not a bug.  It reflects a desire that we use a particular Xtensa HAL layer to save and restore user registers.  The older xtensa code used the HAL, but not this framework (or maybe used it incorrectly).  The newer asm2 layer would need a different framework (e.g. precompile the platform headers to extract the list of TIEs).  But existing boards all work fine, this is about supporting new and out-of-tree ones in a more extensible way.

dcpleung, 2024-02-12:
https://github.com/zephyrproject-rtos/zephyr/pull/67628 has code to save/restore the HiFi registers during context switch. It may be possible to extend that to support arbitrary CP register sets.