zephyrproject-rtos/zephyr #8464: sdk_version file missing

anujdeshpande, 2018-06-19:
```
 $ cmake -GNinja -DBOARD=esp32 ..                                                                                            
CMake Deprecation Warning at /Users/anujdeshpande/code/zephyr/cmake/app/boilerplate.cmake:38 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

-- Found PythonInterp: /Users/anujdeshpande/miniconda3/bin/python3 (found suitable version "3.6.3", minimum required is "3.4")
-- Selected BOARD esp32
Zephyr version: 1.12.99
CMake Error at /Users/anujdeshpande/code/zephyr/cmake/host-tools-zephyr.cmake:24 (message):
  The file '/Users/anujdeshpande/code/zephyr/sdk_version' was not found.  Is
  ZEPHYR_SDK_INSTALL_DIR=/Users/anujdeshpande/code/zephyr misconfigured?
Call Stack (most recent call first):
  /Users/anujdeshpande/code/zephyr/cmake/host-tools.cmake:1 (include)
  /Users/anujdeshpande/code/zephyr/cmake/app/boilerplate.cmake:248 (include)
  CMakeLists.txt:2 (include)

-- Configuring incomplete, errors occurred!
```

- I am trying to get the basic hello world to run - can't seem to find a file called `sdk_version`. As far as I can see there is no such file in the project directory.
- a [comment](https://github.com/zephyrproject-rtos/zephyr/issues/2111#issuecomment-331635596) from #2111 says that there ought to be one.
- Another recent issue says that older project directories did not have the sdk version file ([comment](https://github.com/zephyrproject-rtos/zephyr/issues/5626#issuecomment-356883971) on #5626) - but the present sdk project does not have it.

Is there a script that is supposed to be run that creates this said file?

nashif, 2018-06-19:
Did you 

`export ZEPHYR_TOOLCHAIN_VARIANT=espressif`

?

pswarnak, 2018-06-21:
Hi @anujdeshpande,

It seems to be issue with "ZEPHYR_SDK_INSTALL_DIR" pointing to wrong directory.
Please be aware that Zephyr SDK(Zephyr Software Development Kit - necessary tools and cross-compilers needed to build the kernel on all supported architectures) and [Zephyr RTOS - Project source code](https://github.com/zephyrproject-rtos/zephyr) are two different projects.

Please check the following points:
1.  Is this the correct SDK install directory

> ZEPHYR_SDK_INSTALL_DIR=/Users/anujdeshpande/code/zephyr

2. Try reinstalling SDK by following steps in the link, if required:
http://docs.zephyrproject.org/getting_started/installation_linux.html

3. You additionally need to add/update following variables:
> required to build esp32

export ZEPHYR_TOOLCHAIN_VARIANT=espressif
export ZEPHYR_SDK_INSTALL_DIR="sdk installation directory"

If case, you do not have esp32 toolchain, install it from the [link](https://dl.espressif.com/doc/esp-idf/latest/get-started/index.html#get-started-setup-toolchain) and update the variables properly:
export ESP_IDF_PATH="/path/to/esp-idf"
export ESPRESSIF_TOOLCHAIN_PATH="/path/to/xtensa-esp32-elf/"

anujdeshpande, 2018-06-26:
ZEPHYR_SDK_INSTALL_DIR doesn't seem to be the problem - checked and double checked.
Removing all variables and setting them again worked. Not sure what was the issue.

closed 2018-06-26 by anujdeshpande

nvlsianpu, 2019-07-02:
>ZEPHYR_SDK_INSTALL_DIR doesn't seem to be the problem - checked and double checked.
Removing all variables and setting them again worked. Not sure what was the issue.

I would say that this is transient bug - It appeared for me. Building in new context (another terminal instance) solves the problem.