zephyrproject-rtos/zephyr #8481: Remove the Kconfig helper options for nRF ICs once DT can replace them

carlescufi, 2018-06-19:
The helper options introduced by #8363 are a temporary measure until #7302 is resolved in a way that allows us to handle the equivalent hardware support mapping into ICs properly.

Remove them once the relationship between Kconfig and DT is clarified and there is a clear path ahead on how to deal with instancing.

galak, 2018-12-07:
Not sure if this can be resolved yet or not.

galak, 2019-02-09:
@carlescufi is this resolved?

galak, 2019-10-29:
@carlescufi ?

carlescufi, 2019-11-07:
@galak this is not resolved, the Kconfig options are still [there](https://github.com/zephyrproject-rtos/zephyr/blob/master/soc/arm/nordic_nrf/Kconfig.peripherals). 
@anangl will follow-up after 2.1 code freeze.

nashif, 2021-05-28:
@carlescufi  Any updates on this?

mbolivar-nordic, 2021-05-28:
There is some work in progress in https://github.com/zephyrproject-rtos/zephyr/pull/35398

carlescufi, 2021-06-11:
> There is some work in progress in #35398

Thanks.
@anangl we should try to close this in the 2.7 release cycle.

mbolivar-nordic, 2022-05-25:
@anangl I think this one is finally done, right?

anangl, 2022-05-25:
> @anangl I think this one is finally done, right?

Yes, it is resolved by #44276.

closed 2022-05-25 by anangl