zephyrproject-rtos/zephyr #8497: Need a "monitor" spin-for-ISR API

andyross, 2018-06-20:
Discussion in #8368 showed that some drivers have a need for a "wait for an ISR" behavior in their normally-threaded APIs.  Most of them use a semaphore for this, which works very well.

Nonetheless, when CONFIG_MULTITHREADING=n is set, semaphores and other IPC APIs cannot be used.  Yet this kind of driver behavior may still be required.

One option would be to provide a "monitor" API which can be trivially implemented by a spin loop (but ideally would work via some kind of platform power management feature like x86's MONITOR/MWAIT).

Consider:

```
struct k_monitor { /* arch-specific */ };

/**
 * Busy-wait (ideally in an architecturally-defined low power state)
 * for another CPU or ISR to call k_signal_monitor(), then returns.
 * May return spuriously (for example, after unrelated interrupts) and
 * must be used in a loop.
 */
void k_monitor_wait(struct k_monitor *monitor);

/**
 * Signal any waiters on the address (either on this CPU in a preempted
 * context or on another CPU) to return from k_monitor_wait()
 */
void k_signal_monitor(struct k_monitor *monitor);
```

hongshui3000, 2018-06-22:
@andyross  thanks for your advice
I still have an idea:
Can work_q be enhanced so that it can work in single thread, my driver interrupts the submission of work items, and waits for and processes work items in the main thread

andyross, 2018-06-22:
No.  A work_q **is** a thread.  Really you can't do anything with any of the kernel synchronization primitives when CONFIG_MULTITHREADING=n.  No semaphores, work_q's, fifos/lifos, stacks, msgq's, mailboxes, mutexes.  All of that stuff is predicated on the idea of structuring an application as a bunch of threads that can be blocked to wait on other threads.

All you can do in a single thread is handle interrupts and poll for changes.  It's a very different programming model, not really "Zephyr".

nashif, 2018-06-23:
@ hongshui3000  are you using CONFIG_MULTITHREADING=n with mcuboot?

hongshui3000, 2018-06-23:
No. @nashif   My application generally uses single-threaded, in order to save code and memory usage. Sometimes the benefits of multithreading are not needed due to cost. I generally use only a simple message scheduling mechanism to complete my application. So I hope zephyr will be easier to develop applications in a single thread. Zephyr now supports single-threaded, but development in this case is not easy.If zephyr can provide a complete single-threaded application development framework, it will be of great help to me.

andyross, 2022-06-11:
*(Doing a grooming pass on bugs/issues that are sitting around with my name on or near them.  No one pays me right now so I get to be an unaccountable whirlwind of chaos!)*

Closing.  MULTITHREADING=n keeps going around and around as to what kind of support it wants.  But IMHO it's clear we won't be adding any new synchronization primitives[1]

[1] At least, not until someone wants to rework the backend so {k,sys}_{futex,mutex,sem} can all share the same core code!  There's an issue to that effect somewhere too, and a spin-polled/single-thread lock could be viewed as a feature to that.

closed 2022-06-11 by andyross