zephyrproject-rtos/zephyr #8499: Device Tree support overhaul

carlescufi, 2018-06-20:
This is an umbrella issue for a series of changes to the way Zephyr uses Device Tree.
The following items need to be addressed in order to consider this issue closed:

- [ ] Describe what is the scope of Device Tree. i.e. What should go into Device Tree and what into Kconfig
  - PR
  - Issue
- [ ]  Run Device Tree before Kconfig
  - PR
  - Issue #7302
- [ ] Code generation for mutiple driver instancing
  - PR
  - Issue
- [ ] (optional) Integrate DT support in `menuconfig` 
  - PR
  - Issue

erwango, 2018-06-27:
@SebastianBoe 
> Is the prototype PR supposed to be a part of this effort?

From what we discussed last week, one of  the (/the) first step was to demonstrate the feasibility of using codegen to provide and acceptable solution for driver instantiation (from zephyr user point of view, the end user API). This is what #8561 aims at doing.
This task was supposed to go straight to the point, so I took whatever was available and tried to come up with a demonstrator. If there is a general agreement this is acceptable, then everything should be done the right way.

Doing so, there will be some additional efforts, but at least a plan is there and I hope showing something concrete will help people agreeing this is a desirable thing to have and efforts should be done to achieve the plan. (Rather than endless debates that happened so far in the various PR/issues)

erwango, 2018-09-07:
Added #9406 as bullet of  'Run Device Tree before Kconfig '

erwango, 2018-09-07:
Comment about: " Code generation for instancing of devices ":

Current PR for this point is #8561.
#8561 is based on #6762 that introduces Codegen support. It adds a friendly device declaration module on top of Codegen. It aims at hiding python code from the user in the driver code by providing a nearly python agnostic API.

erwango, 2018-09-11:
Implementation of the current feature could use the edts database proposed in https://github.com/zephyrproject-rtos/zephyr/pull/9876

erwango, 2018-09-11:
Added #9876 as a cut out of #6762 (Codegen introduction)
#9876 focuses on edts database generation that should be used as input for codegen script.
It could also be used as input for "DTS compatibles flags generation" (#9604)

b0661, 2018-09-12:
Sequence of merges for " Code generation for instancing of devices ":

- [ ] Code generation for instancing of devices
1. PR #9876 - creation of extended DTS database using extract_dts_includes.py
2. PR #6762  - inline code generation - codegen 
3. PRs for device driver instantiation by code generation
    - e.g. PR #8561, #9163

galak, 2019-07-30:
edts work handled by PR #17660

mbolivar-nordic, 2020-10-15:
> * Code generation for instancing of devices

Is anyone still advocating for this? I remain unconvinced that we need it and am opposed to the introduction of yet another domain specific language that this would imply for complexity reasons.

> (optional) Integrate DT support in menuconfig

If nobody is still advocating for codegen and the above check box can be fixed, perhaps this can be moved to a separate enhancement and this issue can at last be closed. It isn't doing its job as an umbrella issue since we're not actively using it, and all the other items are checked.

erwango, 2020-10-16:
@mbolivar-nordic Fine with me on both points!

mbolivar-nordic, 2020-10-16:
Great, thanks @erwango. I've filed this issue for a 'menuconfig'-like (really 'guiconfig'-like) interface to devicetree here:

https://github.com/zephyrproject-rtos/zephyr/issues/29295

I think that covers the last unchecked box in this umbrella issue, so let's close it at last and discuss/prioritize additional work elsewhere.

closed 2020-10-16 by mbolivar-nordic

carlescufi, 2020-10-21:
> > * Code generation for instancing of devices
> 
> Is anyone still advocating for this? I remain unconvinced that we need it and am opposed to the introduction of yet another domain specific language that this would imply for complexity reasons.

Not really, but I do recall that there was some talk about whether pinctrl could effectively be implemented **without** code generation. Aside from that, the rest of the usecases (mainly instantiation of the device structures) I think are not justification enough to think about introducing this infrastructure.

erwango, 2020-10-21:
> Not really, but I do recall that there was some talk about whether pinctrl could effectively be implemented without code generation.

@carlescufi Since it is actually available at least on some targets (such as STM32 and Atmel) today, I think we can say this is doable.

carlescufi, 2020-11-13:
> > Not really, but I do recall that there was some talk about whether pinctrl could effectively be implemented without code generation.
> 
> @carlescufi Since it is actually available at least on some targets (such as STM32 and Atmel) today, I think we can say this is doable.

Indeed, and the goal now is  to extend that support to all SoCs in a common manner, thanks for commenting on this @erwango, this just confirms what I already thought.