zephyrproject-rtos/zephyr #8501: I think there is a issue about shell.

xiezhiquan1982, 2018-06-21:
In shell.c file  "get_completion_module" function ,the code "return str ? dest : NULL;" should be "return str ? NULL : dest;"

carlescufi, 2018-06-22:
Assigning to @jhedberg after git blaming

closed 2018-06-22 by nashif

walter-xie, 2018-06-25:
I think line shell.c:522 should be fixed too.

jhedberg, 2018-06-25:
> I think line shell.c:522 should be fixed too.

@walter-xie it seems you're right, but this issue is already closed. You'll need to open a new one. Even better, do you want to try to submit a pull request to fix this second line? This would be an easy task to get started as a contributor to the Zephyr project.

jhedberg, 2018-07-07:
>> I think line shell.c:522 should be fixed too.

> @walter-xie it seems you're right, but this issue is already closed. You'll need to open a new one. Even better, do you want to try to submit a pull request to fix this second line? This would be an easy task to get started as a contributor to the Zephyr project.

It's now a couple of weeks since the above comment, and I haven't seen any response or new pull request, so I went ahead and created one myself: #8795

PR #8537 (2018-06-22): shell: Fix command completion logic
The original code (introduced by commit d5db35204a4) looked like this before
the last rewrite/cleanup:

       if (default_module != -1) {
               return (str == NULL) ? dest : -1;
       }

However with the cleanup the logic seems to have gotten reversed.

Fixes #8501
commit b26ca136 (2018-06-22): shell: Fix command completion logic

The original code (introduced by commit d5db35204a4) looked like this
before the last rewrite/cleanup:

       if (default_module != -1) {
               return (str == NULL) ? dest : -1;
       }

However with the cleanup the logic seems to have gotten reversed.

Fixes #8501