zephyrproject-rtos/zephyr #8506: tests/subsys/fs/fat_fs_api - test_fat_mount results into assertion failure on Arduino_101 - FS init failed (-19)

pswarnak, 2018-06-21:
The test **tests/subsys/fs/fat_fs_api** results in Assertion failure in test_fat_mount sub-test case. 

**Arch:** x86
**Board:** arduino_101
**Zephyr commit:** c738834854849adad705dee38770aef61bd1b4a2.

**Error Console Log:**
```
***** Booting Zephyr OS v1.12.0-284-gc738834 *****
Running test suite fat_fs_basic_test
===================================================================
starting test - test_fat_mount
ASSERTION FAIL [(res == FR_OK)] @ zephyr/subsys/fs/fat_fs.c:333:
        FS init failed (-19)
```

**Steps to reproduce:**
```
cd zephyr/tests/subsys/fs/fat_fs_api/
rm -rf build && mkdir build && cd build
cmake -D BOARD=arduino_101 ../
make BOARD=arduino_101 flash

Check the console log.
```

pswarnak, 2018-06-22:
**Additional information:**
As per "git bisect", the commit resulting into this test failure is 2fe51996db89c3e941134178b50c2c6982f9b5c7 by @tbursztyka.
By reverting the mentioned commit, test could pass as expected.

ramakrishnapallala, 2018-07-20:
@pswarnak  can you update the status? is it same as before?

pswarnak, 2018-07-23:
Hi @ramakrishnapallala,

The test still fails on ARDUINO_101/x86 with the same error signature:

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.9.0-rc1-6413-gef93589 (delayed boot 1000ms) *****
Running test suite fat_fs_basic_test
===================================================================
starting test - test_fat_mount
ASSERTION FAIL [(res == FR_OK)] @ /home/pswarnak/workspace/1.13_zephyr/subsys/fs/fat_fs.c:333:
        FS init failed (-19)
```

ramakrishnapallala, 2018-07-24:
@tbursztyka can you look into this issue ?

nashif, 2018-08-28:
@SavinayDharmappa can you please take a look at this while @ramakrishnapallala and @tbursztyka are out?

SavinayDharmappa, 2018-08-29:
https://github.com/zephyrproject-rtos/zephyr/pull/9714

PR #9714 (2018-08-29): boards: x86: arduino_101: set DISK_FLASH_MAX_RW_SIZE to 256
set DISK_FLASH_MAX_RW_SIZE to 256, as maximum bytes that can be
programed to SPI FLASH by Page Program Command is 256.

Fixes #8506
commit 8c91001a (2018-08-29): boards: x86: arduino_101: set DISK_FLASH_MAX_RW_SIZE to 256

set DISK_FLASH_MAX_RW_SIZE to 256, as maximum bytes that can be
programed to SPI FLASH by Page Program Command is 256.

Fixes #8506
closed 2018-08-29 by nashif