zephyrproject-rtos/zephyr #8529: tests/kernel/common/kernel.common fails for native_posix on Ubuntu 16.04

apurbochakrabarty1, 2018-06-22:
The failure occurs while running sanitycheck when native_posix builds with the host compiler (gcc 5.4.0 20160609) on Ubuntu 16.04 (Ubuntu 5.4.0-6ubuntu1~16.04.9) . 

The tests/kernel/common/kernel.common fails as follows:

```[ 34%] Building C object CMakeFiles/app.dir/src/main.c.obj
ccache /usr/bin/gcc -DBUILD_VERSION=v1.12.0-290-g7a7e4f5 -DKERNEL -D_POSIX_C_SOURCE=200809 -D__ZEPHYR__=1 -I/ho
/home/apurboch/zephyr/tests/kernel/common/src/intmath.c:50:1: error: bit-field ‘__error_if_negative’ width not
 NEG_CHECK(int, 0x80000000);
 ^
/home/apurboch/zephyr/tests/kernel/common/src/intmath.c:52:1: error: bit-field ‘__error_if_negative’ width not
 ROLLOVER_CHECK(int, 2147483647, -2147483648);
 ^
cc1: error: unrecognized command line option ‘-Wno-format-truncation’ [-Werror]
cc1: all warnings being treated as errors
CMakeFiles/app.dir/build.make:140: recipe for target 'CMakeFiles/app.dir/src/intmath.c.obj' failed
make[3]: *** [CMakeFiles/app.dir/src/intmath.c.obj] Error 1```

aescolar, 2018-06-23:
Indeed..
@andyross , you added those checks (e9174c6ddeddf58d8aebba5c89e50965731899e6), and seems you were trying to fix a similar issue in 
718a4cd24597853c9151ff57978a0b3de05c5329 So, would you mind taking it?

The exact error:
```
In file included from /usr/include/features.h:367:0,
                 from /usr/include/stdint.h:25,
                 from /usr/lib/gcc/x86_64-linux-gnu/5/include/stdint.h:9,
                 from [..]/include/zephyr/types.h:10,
                 from [..]/include/kernel_includes.h:17,
                 from [..]/include/kernel.h:17,
                 from [..]/include/zephyr.h:18,
                 from [..]/tests/ztest/include/ztest.h:47,
                 from [..]/tests/kernel/common/src/intmath.c:8:
[..]/include/toolchain/gcc.h:26:28: error: bit-field ‘__error_if_negative’ width not an integer constant
 #define BUILD_ASSERT(EXPR) _Static_assert(EXPR, "")
                            ^
[..]/tests/kernel/common/src/intmath.c:24:25: note: in expansion of macro ‘BUILD_ASSERT’
 #define NEG_CHECK(T, N) BUILD_ASSERT((-((T)N)) == (~((T)N))+1)
                         ^
[..]/tests/kernel/common/src/intmath.c:50:1: note: in expansion of macro ‘NEG_CHECK’
 NEG_CHECK(int, 0x80000000);
 ^
[..]/include/toolchain/gcc.h:26:28: error: bit-field ‘__error_if_negative’ width not an integer constant
 #define BUILD_ASSERT(EXPR) _Static_assert(EXPR, "")
                            ^
[..]/tests/kernel/common/src/intmath.c:27:37: note: in expansion of macro ‘BUILD_ASSERT’
 #define ROLLOVER_CHECK(T, MAX, MIN) BUILD_ASSERT((T)((T)1 + (T)MAX) == (T)MIN)
                                     ^
[..]/tests/kernel/common/src/intmath.c:52:1: note: in expansion of macro ‘ROLLOVER_CHECK’
 ROLLOVER_CHECK(int, 2147483647, -2147483648);
 ^
```
Seems to happen when setting this gcc version to c99.
Setting it to 2011 it works just fine.
```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 1c9eb62..7696310 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -120,7 +120,7 @@ zephyr_compile_options(
 )
 
 zephyr_compile_options(
-  $<$<COMPILE_LANGUAGE:C>:-std=c99>
+  $<$<COMPILE_LANGUAGE:C>:-std=c11>
 
   $<$<COMPILE_LANGUAGE:CXX>:-std=c++11>
   $<$<COMPILE_LANGUAGE:CXX>:-fcheck-new>
```

Also it seems that before your fix in 718a4cd24597853c9151ff57978a0b3de05c5329 it was also breaking, just a bit differently:
```
[..]/tests/kernel/common/src/intmath.c:50:30: error: enumerator value for ‘__build_assert37’ is not an integer constant
[..]/tests/kernel/common/src/intmath.c:52:30: error: enumerator value for ‘__build_assert41’ is not an integer constant
 ROLLOVER_CHECK(int, 2147483647, -2147483648);
                              ^
```

aescolar, 2018-06-23:
Related to #8093

aescolar, 2018-06-24:
#8520 will most likely fix this as a sideeffect of changing to C 2011

andyross, 2018-06-25:
Ugh, this is an ugly one.  At least a proximate cause is a host tooling glitch.  GCC 5.4 supports _Static_assert() just fine, but it's not using it on Ubuntu 16.04.

Instead, there is code in /usr/include/x86_64-linux-gnu/sys/cdefs.h that looks like this:

```
#if (!defined _Static_assert && !defined __cplusplus \
     && (defined __STDC_VERSION__ ? __STDC_VERSION__ : 0) < 201112 \
     && (!__GNUC_PREREQ (4, 6) || defined __STRICT_ANSI__))
# define _Static_assert(expr, diagnostic) \
    extern int (*__Static_assert_function (void)) \
      [!!sizeof (struct { int __error_if_negative: (expr) ? 2 : -1; })]
#endif
```

This is a glibc file, pulled in transitively from stdint.h.  And if you follow the logic, it will refuse to use the compiler builtin when we're in __STRICT_ANSI__ and not at a C11+ standard level (which seems wrong to me: shouldn't that be only when -pedantic is passed?  Normally gcc does *not* disable extensions in -std=... arguments).

And it fails. I actually don't know why it fails.  It works on all but two of hte expressions, and those definitely result in compile-time integer constants.  Maybe there's a limit of expression depth or some kind of other interaction with the 32 bit rollover being tested?  It definitely smells like a compiler bug: gcc 8.1.1 on Fedora hits exactly the same path building native_posix (i.e. it uses the glibc macro and not the builtin) and builds correctly.

So... options:

1. We can pass -std=c11 in native_posix.  This would be my preference, though I don't know if we have policies about this and it sounds like you tried it and it resulted in other failures somewhere else? (This one intmath.c file builds with with it, at least).

2. We can add `#define _Static_assert _Static_assert` before the system includes in this one test file, which trips the first part of that test.  Awful hack, but will change fewer things and involves less risk.

2b. We can add -D_Static_assert=_Static_assert to the command line for native_posix globally.  Because we really don't want that native macro, we want the builtin!  Less of a hack.  Somewhat uglier.  Hard to document cleanly.

3. We can disable the two relevant expressions test on this one platform combination (Ubuntu + native_posix).  Though we'll probably rediscover the same issue somewhere else.

andyross, 2018-06-25:
Back to @aescolar as the fix requires policy decisions internal to native_posix and I don't know what to pick :)

andrewboie, 2018-06-27:
we keep tripping over issues with host GCC version and the native build.
maybe the best solution is to just put a specific version of the host compiler in the SDK?

aescolar, 2018-06-27:
@andyross : I have been stalling a bit this one, expecting the llvm one would make it in with the c11 switch (which should just work for the native port)
@andrewboie : That file seems to be from libc6, not from gcc, so I guess bundling the compiler would not have helped in this case. Anyhow, the only real compiler issue I recall was due to new warnings (#7610), and those may actually reveal something we want to fix, and if not we could always add a pragma. So, I guess we do not need to give up on supporting whatever gcc version the host has (yet).

aescolar, 2018-07-02:
@andyross I think it would be best to go for C 2011. _Static_assert being a 2011 feature, even if most compilers support it with older versions, it would seem we cannot rely on them doing it correctly.
The other options you proposed seem to be too specific hacks for this particular compiler+c library version, apart from too hackish.

PR #8749 (2018-07-05): For native_posix set the compiler to C standard 2011
Allow archs/boards/apps to override the C standard version.
+
For native_posix, override the C standard used for compilation to C 2011
This is due to some tests using _Static_assert which is a 2011 feature, but otherwise relying on compilers supporting it also when set to C99.
This was in general ok, but with some host compilers and C library versions it led to problems. So we override it to 2011 for native_posix board.

Fixes: #8529
commit 0a5de7df (2018-07-05): native_posix: override C standard version to 2011

Override the C standard used for compilation to C 2011
This is due to some tests using _Static_assert which is a 2011
feature, but otherwise relying on compilers supporting it also
when set to C99.
This was in general ok, but with some host compilers and C
library versions it led to problems. So we override it to 2011
for native_posix board.

Fixes: #8529
closed 2018-07-05 by nashif