zephyrproject-rtos/zephyr #8539: Categorise Kconfig options in documentation

nashif, 2018-06-23:
symbol reference in http://docs.zephyrproject.org/reference/kconfig/index.html is too long and has every single Kconfig defined in the system. We should split this based on categories and subsystems, for example group all kernel Kconfigs in one section, drivers all in one section, networking, bluetooth, etc.

This should make it easy to find config options and makes the documentation of symbols more usable.

nashif, 2018-06-23:
@ulfalizer fyi, what do you think?

ulfalizer, 2018-06-23:
Might be tricky to split it up logically. Depends on everything related to X being under some top-level X menu or similar.

Could split it into a few pages though, with e.g. max 1000 symbols per page, and have a top-level index page with A-F, G-L, etc., links. That'd be nicer to the browser at least (though I still think it's handy to have it all on one page sometimes when it works).

ulfalizer, 2018-06-27:
@nashif 
Should be able to work something out at least. Just gotta figure the best way to organize it. Could do it by Kconfig structure, directory structure, etc.

Might be overkill to mirror the entire Kconfig tree I think. What people are the most familiar with what different subsystems there are, etc.? They might have some opinions. I've mostly dealt with the Kconfigs at a higher level.

dbkinder, 2018-07-17:
An approach would be to generate top-level reST files for each of the major kconfig areas, as defined in the top-level /Kconfig.zephyr file:
    
- source "arch/Kconfig"
- source "kernel/Kconfig"
- source "dts/Kconfig"
- source "drivers/Kconfig"
- source "misc/Kconfig"
- source "lib/Kconfig"
- source "subsys/Kconfig"
- source "ext/Kconfig"
- source "tests/Kconfig"

and still generate the 3800+ option-specific files as we do today, but break up the index.rst into these nine files and change the current index.rst into just a toctree to these nine section files.

ulfalizer, 2019-12-04:
Categorized by path now that https://github.com/zephyrproject-rtos/zephyr/pull/20322 is in.

Could probably be tweaked further, but closing now that the support is there.

closed 2019-12-04 by ulfalizer