zephyrproject-rtos/zephyr #8563: Compilation warning/errot on stm32l4: "__weak" redefined

cpriouzeau, 2018-06-26:
After the introduction of patch aee97be7107 : "arch: arm: soc: remove core zephyr header inclusions from soc.h", some error appear at compilation time:

In file included from /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/include/toolchain.h:21:0,
                 from /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/include/kernel_includes.h:19,
                 from /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/arm/soc/st_stm32/stm32l4/soc.h:26,
                 from /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/arm/soc/st_stm32/common/arm_mpu_regions.c:7:
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/include/toolchain/gcc.h:145:0: error: "__weak" redefined [-Werror]
 #define __weak __attribute__((__weak__))

To solve it, on stm32l4 and stm32f4 we need to move the call of "include <kernel_includes.h>" before machine specific call "include <stm32l4xx.h>" or "include <stm32f4xx.h>"

Proposal to correct it with stm32l4 and smt32f4:
<pre>
diff --git a/arch/arm/soc/st_stm32/stm32f4/soc.h b/arch/arm/soc/st_stm32/stm32f4/soc.h
index 044701f..8e462bf 100644
--- a/arch/arm/soc/st_stm32/stm32f4/soc.h
+++ b/arch/arm/soc/st_stm32/stm32f4/soc.h
@@ -24,9 +24,8 @@
 
 #ifndef _ASMLANGUAGE
 
-#include <stm32f4xx.h>
-
 #include <kernel_includes.h>
+#include <stm32f4xx.h>
 
 #ifdef CONFIG_CLOCK_CONTROL_STM32_CUBE
 #include <stm32f4xx_ll_utils.h>
diff --git a/arch/arm/soc/st_stm32/stm32l4/soc.h b/arch/arm/soc/st_stm32/stm32l4/soc.h
index fafe672..025c370 100644
--- a/arch/arm/soc/st_stm32/stm32l4/soc.h
+++ b/arch/arm/soc/st_stm32/stm32l4/soc.h
@@ -22,8 +22,8 @@
 #ifndef _ASMLANGUAGE
 
 #include <autoconf.h>
-#include <stm32l4xx.h>
 #include <kernel_includes.h>
+#include <stm32l4xx.h>
 
 #define GPIO_REG_SIZE         0x400
 /* base address for where GPIO registers start */
</pre>

galak, 2018-06-28:
Proper fix will be a combo of #8616 and #8618

ioannisg, 2018-06-29:
Closing this, again, since #8616 and #8618 are merged in.

commit 30529da0 (2018-06-27): arch: arm: stm32: correct include issue

This commit correct the issue introduce by aee97be which
changed the order of header inclusion that lead
to warning generation at compilation, that truend into errors
in CI.
This commit reinstantiate the initial header inclusion order.

Fixes #8563
closed 2018-06-29 by ioannisg