zephyrproject-rtos/zephyr #8566: cmake: Allow setting BOARD_ROOT from the enviroment

aescolar, 2018-06-26:
It would be nice if it were possible to set BOARD_ROOT from the environment, and have cmake use it when building an application. Similarly to how it already accepts BOARD being set in this manner.

nashif, 2019-02-21:
afaik, this is already implemented.

SebastianBoe, 2019-02-22:
It is not unfortunately.

```
/home/sebo/ncs/zephyr/cmake/app/boilerplate.cmake
248:# 'BOARD_ROOT' is a prioritized list of directories where boards may
250:list(APPEND BOARD_ROOT ${ZEPHYR_BASE})
267:foreach(root ${BOARD_ROOT})

/home/sebo/ncs/zephyr/cmake/usage/CMakeLists.txt
1:string(REPLACE ";" " " BOARD_ROOT_SPACE_SEPARATED "${BOARD_ROOT}")
7:  -DBOARD_ROOT_SPACE_SEPARATED=${BOARD_ROOT_SPACE_SEPARATED}

/home/sebo/ncs/zephyr/cmake/usage/usage.cmake
16:string(REPLACE " " ";" BOARD_ROOT "${BOARD_ROOT_SPACE_SEPARATED}")
20:  foreach(root ${BOARD_ROOT})

/home/sebo/ncs/zephyr/cmake/extensions.cmake
1132:  string(REPLACE ";" " " BOARD_ROOT_SPACE_SEPARATED "${BOARD_ROOT}")
1137:    -DBOARD_ROOT_SPACE_SEPARATED=${BOARD_ROOT_SPACE_SEPARATED}

/home/sebo/ncs/zephyr/samples/application_development/out_of_tree_board/CMakeLists.txt
4:set(BOARD_ROOT ${CMAKE_CURRENT_LIST_DIR})

```

aescolar, 2019-02-22:
I just marked it as low, as what we have today does enable the use case (even if it would be a bit nicer with this option)

nordicjm, 2024-11-19:
Supported in newer versions of zephyr

closed 2024-11-19 by nordicjm