zephyrproject-rtos/zephyr #8567: Can't parse json

chtavares592, 2018-06-26:
Hello, I'm developing a simple app to parse a JSON. That JSON is an array of object's array that format is: `{"products":[[{"test":"1","mode":"testing"}],[{"test":"2","mode":"testing"}]]}`  
I already verify if this json is valid in this website https://jsonlint.com/ and is all ok.
Bellow is how I describe this json:

```
struct data_recv_products {
	const char *mode;
	const char *test;
};

struct data_recv_products_array {
	struct data_recv_products items[4];
	size_t items_len;
};

struct data_recv {
	struct data_recv_products_array products[4];
	size_t products_len;
};

static const struct json_obj_descr data_recv_products_descr[] =
{
	JSON_OBJ_DESCR_PRIM(struct data_recv_products,
			    mode, JSON_TOK_STRING),
	JSON_OBJ_DESCR_PRIM(struct data_recv_products,
			    test, JSON_TOK_STRING),
};

static const struct json_obj_descr data_recv_products_array_descr[] =
{
	JSON_OBJ_DESCR_OBJ_ARRAY(struct data_recv_products_array, items, 4,
				 items_len, data_recv_products_descr,
				 ARRAY_SIZE(data_recv_products_descr)),
};

static const struct json_obj_descr data_recv_descr[] =
{
	JSON_OBJ_DESCR_OBJ_ARRAY(struct data_recv, products, 4,
				 products_len, data_recv_products_array_descr,
				 ARRAY_SIZE(data_recv_products_array_descr)),
};
```
 However when I call the json parse function returns -einval. I was debugging and I found where occurs the problem, is in equivalent_types on decode_value function at json.c. Seems the function wait for "{" and received the "[". 
I don't have sure if I build the json's descriptors right or the json's library on zephyr doesn't offer support for that format of json. 
Here is the link for my repository test:  https://github.com/chtavares592/json_test/
Could someone help me? Thanks!

lpereira, 2018-06-27:
I spent some time debugging this.  Things seem fine in the JSON library, but I've never the use case for nested arrays, so it's taking me more time than I imagined to make this work.

In order to make this work, one would need another descriptor macro such as this one:

```C
#define JSON_OBJ_DESCR_ARRAY_ARRAY(struct_, field_name_, max_len_, len_field_, \
                                   elem_descr_, elem_descr_len_)        \
  {        \
    .field_name = (#field_name_), .field_name_len = sizeof(#field_name_) - 1,  \
    .offset = offsetof(struct_, field_name_),        \
    .alignment = __alignof__(struct_) - 1, .type = JSON_TOK_LIST_START,        \
    .array = {        \
        .element_descr =        \
            &(struct json_obj_descr){        \
                .type = JSON_TOK_LIST_START,        \
                .object =        \
                    {        \
                        .sub_descr = elem_descr_,        \
                        .sub_descr_len = elem_descr_len_,        \
                    },        \
                .offset = offsetof(struct_, len_field_),        \
                .alignment = __alignof__(struct_) - 1,        \
            },        \
        .n_elements = (max_len_),        \
    },        \
  }
```

And use it to define `json_data_recv_descr[]` instead.  This almost works, after making a lot of changes to `json.c` to add debugging calls to `printk()`, but still fails expecting a `[` where a `{` would make sense.

I haven't debugged any further, and unfortunately won't be able to spend much more time on this.  I hope this is sufficient to help you debug this, though.

chtavares592, 2018-06-28:
Thanks @lpereira for answering me, I will try developing a macro for my use based on your example.

chtavares592, 2018-06-28:
@lpereira finally I can parser that JSON format, I used your macro and change my second descriptor of object array for to just object and works. If you want to see here is my directory: https://github.com/chtavares592/json_test
Thanks for all help.

closed 2018-06-28 by chtavares592

lpereira, 2018-06-28:
Glad it worked for you!