zephyrproject-rtos/zephyr #8584: ToolchainCapabilityDatabase.cmake:93 error in PR #8579

avisconti, 2018-06-27:
Please help me. In submitting PR #8579 the automatic checks provides an error which I guess it is not related to the PR iself. I noticed a similar issue was found and solved with #7514 on May 14.

Details:
CMake Error at /home/buildslave/.cache/zephyr/ToolchainCapabilityDatabase.cmake:93:
  Parse error.  Function missing ending ")".  End of file reached.
Call Stack (most recent call first):
  ../../../cmake/extensions.cmake:638 (include)
  ../../../cmake/extensions.cmake:1020 (zephyr_check_compiler_flag)
  ../../../cmake/extensions.cmake:996 (target_cc_option_fallback)
  ../../../cmake/extensions.cmake:106 (target_cc_option)
  ../../../CMakeLists.txt:212 (zephyr_cc_option)

Full log can be found here:
https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/17843/3/console

@SebastianBoe @nashif

SebastianBoe, 2018-06-27:
What it's saying is that this file is corrupted:

/home/buildslave/.cache/zephyr/ToolchainCapabilityDatabase.cmake

Without this file it is impossible to debug.

avisconti, 2018-06-27:
@SebastianBoe I saw you wanted to recheck the CI on #8579 
Is there something I can do?

SebastianBoe, 2018-06-27:
That was a command to CI, I'm not sure it worked though.

I wanted to find out if this happened every time or if it was spurious.

If it is spurious we can accept this until it happens again, if it happens every time we need to have CI print the file somehow to debug.

SebastianBoe, 2018-06-27:
@nashif : Is the 'recheck' functionality down? Is there any other way to re-run CI?

avisconti, 2018-06-27:
I could repush the commit with a negligble change

avisconti, 2018-06-27:
@SebastianBoe 
Done! Let's wait...

SebastianBoe, 2018-06-27:
Right, you are going to have to find some way of catting the file then. Either from sanitycheck.py or .shippable, it's not possible from CMake I believe.

In hindsight, some config file format should have been used for the database. CMake can't do error-checking  on syntax errors in the CMake code itself ...

avisconti, 2018-06-28:
@SebastianBoe 
> Right, you are going to have to find some way of catting the file then. 
> Either from sanitycheck.py or .shippable, it's not possible from CMake I believe.

Not sure I'll be able to do that easily, as I do not know the CI scripting system much...

Anyway, I re-ran it with latest commits once more. If it fails again I'll try first to run it locally.
Is it easy to do that? I think I'll just need to run ./scripts/sanitycheck right?

avisconti, 2018-06-28:
@SebastianBoe 
Mmmmh ... now sanitycheck worked.
The reason why I re-ran it was because I noticed this recent patch and I rebased my changed 
on top of it:
--
Author: Andrew Boie <andrew.p.boie@intel.com>
Date:   Tue Jun 26 10:41:35 2018 -0700

    sanitycheck: fix spammy build output
--

Do you think this might be the reason why it is now working?

SebastianBoe, 2018-06-28:
I doubt it.

My best guess would be that corruption can occur when there are multiple consumers and producers. If this happens again I'll re-write this to use a config format that can be validated.

avisconti, 2018-06-28:
In fact I then realized that the commit was already present even before, when the sanitycheck failed.
Anyway, If you agree I'll close this bug item.

closed 2018-06-28 by SebastianBoe