zephyrproject-rtos/zephyr #8602: master broken for many arm boards

aescolar, 2018-06-28:
30529da0e4ab6ff967ccdd18dcae4f161de7be24 broke master for quite many arm boards.
See for example https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/17898/summary/console
Can be reproduced easily by running locally, for example:
```sanitycheck -l -T tests/lib/sprintf/```
```
96b_carbon                tests/lib/sprintf/libraries.libc                   FAILED: build_error
disco_l475_iot1           tests/lib/sprintf/libraries.libc                   FAILED: build_error
nucleo_f411re             tests/lib/sprintf/libraries.libc                   FAILED: build_error
nucleo_l476rg             tests/lib/sprintf/libraries.libc                   FAILED: build_error
stm32f411e_disco          tests/lib/sprintf/libraries.libc                   FAILED: build_error
nucleo_f412zg             tests/lib/sprintf/libraries.libc                   FAILED: build_error
nucleo_f429zi             tests/lib/sprintf/libraries.libc                   FAILED: build_error
nucleo_f401re             tests/lib/sprintf/libraries.libc                   FAILED: build_error
olimex_stm32_e407         tests/lib/sprintf/libraries.libc                   FAILED: build_error
olimex_stm32_h407         tests/lib/sprintf/libraries.libc                   FAILED: build_error
stm32f429i_disc1          tests/lib/sprintf/libraries.libc                   FAILED: build_error
olimex_stm32_p405         tests/lib/sprintf/libraries.libc                   FAILED: build_error
stm32f469i_disco          tests/lib/sprintf/libraries.libc                   FAILED: build_error
nucleo_f413zh             tests/lib/sprintf/libraries.libc                   FAILED: build_error
stm32f412g_disco          tests/lib/sprintf/libraries.libc                   FAILED: build_error
stm32f4_disco             tests/lib/sprintf/libraries.libc                   FAILED: build_error
nucleo_f446re             tests/lib/sprintf/libraries.libc                   FAILED: build_error
total complete:  102/ 102  100%  failed:   17
85 of 102 tests passed with 0 warnings in 141 seconds
```

aescolar, 2018-06-28:
CC @nashif

erwango, 2018-06-28:
Same as #8593

aescolar, 2018-06-28:
> Same as ...

@erwango : Uhm, ok, that one referred to another commit, and there does not seem to be an issue in 30529da0e4ab6ff967ccdd18dcae4f161de7be24^ == b312762

galak, 2018-06-28:
Working on it.

galak, 2018-06-28:
Closing since this is covered by #8593 & #8563

closed 2018-06-28 by galak