zephyrproject-rtos/zephyr #8610: USB: Setup stage in control transfers

pawelzadrozniak, 2018-06-28:
Hello.
I have noticed that the setup stage in USB OUT control transfers is handled in stack (which is OK), but in case of IN transfers it's done by the driver.

Example:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_device.c
host-to-device: lines 295 and 331
device-to-host (i.e. GetDescriptor): line 336, no setup stage (OUT transaction with no data)

example on stm32: https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_stm32.c
line 648, status stage (OUT, no data).

With current implementation, longer transfers (with multiple INs in a row) are impossible (i.e. long descriptors, let's say for HID). Driver has no information whether the current _write() is the last one if length parameter is equal to EP size (i.e. 64). This will cause device-to-host control transfers to fail **if data length is a multiple of EP size (i.e. DFU upload - one IN transaction, 64B).** The device stack should be responsible for handling the status stage.

I'd like to propose a change - move status stage to the stack.
Any comments / objections?

jfischer-phytec-iot, 2018-06-28:
As far as I can remember it is handled in the stack, driver should not have to do with it, I would not take a hal driver as a reference. There is an issue that some parts of Setup Stage could run in the ISR context, see e.g. https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_kinetis.c#L943

pawelzadrozniak, 2018-06-29:
Take a look again at usb_device.c - usb_handle_control_transfer().
Status is sent explicitly in host-to-device control transfer only (295, 331).
In case of device-to-host, only the data packets are send and there is no following OUT transfer (usb_dc_ep_read()) (or am I missing something?).

This transaction is missing:
[image]

jfischer-phytec-iot, 2018-07-02:
The stack handles Status OUT ACK in [usb_device.c L:299](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_device.c#L297)

>We have a special case (nRF52840), where the setup stage is handled by hardware and triggered differently than standard OUT transfer.

It is about Control read transfer as shown in Figure 179, right? The automatism is only possible by activate shorts EP0DATADONE->EP0STATUS, or I overlook something? Rises the controller a event when status stage is over, so when Status OUT ACK has arrived? Do you have to use EP0STATUS? Can not one simply put EP0 in RX mode after EP0DATADONE?

To consider different types of automatism, like address assignment or setup stage, we can implement a function to query different capabilities of the controllers and handle them in the stack accordingly.

pawelzadrozniak, 2018-07-02:
Ok, now I see.

In case of nRF52840, triggering EP0STATUS after all of INs is required to complete the transaction and be able to handle next setup packet. Zero-length packet in status stage is fully handled by hardware and software does not get any notification.
It also has a second role - it will prevent host from sending any more data (by automatic stalling) after it was triggered. Anyway, action must be taken after last IN was completed - software will not be notified about the following OUT (ZLP).

Something like this:
https://github.com/pawelzadrozniak/zephyr/commit/d4c9c3627398351dda8ab10337267f7ca235884e#diff-609952c961f3c77cb7811bedee4bfd2cR336

jfischer-phytec-iot, 2018-07-03:
>Anyway, action must be taken after last IN was completed - software will not be notified about the following OUT (ZLP).

Sounds like a bug :smile:. In this case I suggest not to change the stack, but evaluate the event EP0DATADONE (or shortcut EP0DATADONE->EP0STATUS) and then trigger (?EP0STATUS task and) the ep0 out callback in the driver.

How far are you with the redesign, can you open a PR?

pawelzadrozniak, 2018-07-04:
That's how the hardware was designed - status is handled by hardware so it does not need to notify the software. EP0STATUS must be triggered instead - either by enabling the short or explicitly writing the task register.

The problem is elsewhere - take a look at this graph:  http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.nrf52840.ps/ip/usbd/doc/image/ctrl_read.svg .

If there is more subsequent INs, then EP0DATADONE will occur multiple times. Driver must know which IN transaction is the last one in series (when to enable EP0DATADONE->EP0STATUS short or "manually" trigger EP0STATUS).
Stack handles the fragmentation (in this case - 64-byte chunks) and performs multiple writes, i.e. 100-byte descriptor will be split into 64 and 36-byt packet. If write length is smaller than 64 (last packet) - then that is the last IN transaction and EP0STATUS can be triggered. However, if the initial data length is a multiple of 64 then driver does not know of this is the last chunk and if it can proceed to status stage.

I have a working prototype and should be able to submit a PR this week.

jfischer-phytec-iot, 2018-07-04:
>If there is more subsequent INs, then EP0DATADONE will occur multiple times. Driver must know which IN transaction is the last one in series (when to enable EP0DATADONE->EP0STATUS short or "manually" trigger EP0STATUS).

The driver could track the number of IN packets and compare them with wLength. wLength is accessible to the driver because of the design of the controller. Is it a possible and acceptable solution?

pawelzadrozniak, 2018-07-04:
In case of control read (i.e. GetDescriptor) host sends setup packet maximum acceptable length as wLength (i.e. 255) and the actual data length is chosen by software (i.e. descriptor length). The needed value is stored in usb_dev.data_buf_len, but it's not available to the driver.

https://github.com/pawelzadrozniak/zephyr/blob/master/subsys/usb/usb_device.c#L292

jfischer-phytec-iot, 2018-07-04:
Other suggestion:
```
--- a/subsys/usb/usb_device.c
+++ b/subsys/usb/usb_device.c
@@ -234,7 +234,8 @@ static void usb_data_to_host(void)
        u32_t chunk = min(MAX_PACKET_SIZE0, usb_dev.data_buf_residue);
 
        /*Always EP0 for control*/
-       usb_dc_ep_write(0x80, usb_dev.data_buf, chunk, &chunk);
+       usb_dc_ep_write(0x80, usb_dev.data_buf, usb_dev.data_buf_residue,
+                       &chunk);
        usb_dev.data_buf += chunk;
        usb_dev.data_buf_residue -= chunk;
 }

```
The driver should transfer as many bytes as it can, so most simply MPS for the corresponding endpoint.

pawelzadrozniak, 2018-07-05:
That seems to work fine.

nashif, 2018-07-31:
it works, so lets close it.

closed 2018-07-31 by nashif