zephyrproject-rtos/zephyr #8611: RT1050EVK: MPU FAULT with Zephyr OS v1.12.0-360-gf3d1b22 using ztest

florinleotescu, 2018-06-28:
### Issue description
Using ztest function (code bellow) with the **CONFIG_STDOUT_CONSOLE=y** in **prj.conf** generates an MPU FAULT. This happens with the current version of zephyr (v1.12.0), however was not observed in zephyr version 1.11.99.

### Code used:
```
#include <ztest.h>

void my_test(void)
{
}
void test_main(void)
{
          ztest_test_suite(test_tests, ztest_unit_test(my_test));
          ztest_run_test_suite(test_tests);
}
```
### Ouput
```
***** Booting Zephyr OS v1.12.0-360-gf3d1b22 *****
***** MPU FAULT *****
  Executing thread ID (thread): 0x200008ac
  Faulting instruction address:  0x0
  Stacking error
  Data Access Violation
  Address: 0x2000181c
Fatal fault in essential thread! Spinning...
```

MaureenHelm, 2018-06-28:
You're probably blowing up the stack. We enabled hardware stack checking by default for tests in Zephyr 1.12, and subsequently found a number of tests that had always been quietly blowing up the stack .

CONFIG_STDOUT_CONSOLE=y increases stack usage, so we disabled it for some testcases. See commit aa26d99

florinleotescu, 2018-07-02:
Oh, ok! I missed that using CONFIG_ZTEST=y the default stack value is 512. 
I increased the stack size (1024) using CONFIG_MAIN_STACK_SIZE and it works.

closed 2018-07-02 by florinleotescu