zephyrproject-rtos/zephyr #8614: cmake: Zephyr wrapped functions does not allow keywords on zephyr_link_libraries

tejlmand, 2018-06-28:
During work on: #8451 I discovered that the wrapping macro:  zephyr_library_named()
break the possibility of later using PUBLIC / PRIVATE / INTERFACE keywords together with
zephyr_library_link_libraries(INTERFACE <linking_flag>)
or using plain cmake:
target_link_libraries(myLib INTERFACE <linking_flag>)

as cmake will break with:
`The plain signature for target_link_libraries has already been used with`
`the target "zephyr".  All uses of target_link_libraries with a target must`
`be either all-keyword or all-plain.`

`The uses of the plain signature are here:`

`  * ../../../cmake/extensions.cmake:431 (target_link_libraries)`

SebastianBoe, 2018-06-28:
Thank you for reporting.

This has been fixed. The fix will be up for review after all of it's dependent PR's are merged:

Dependents:
https://github.com/zephyrproject-rtos/zephyr/pull/8533

Branch with fix:
https://github.com/SebastianBoe/zephyr/tree/proper_fix_of_8438

tejlmand, 2018-06-29:
Hi,

your fix does not fix the problem, only moves it around, which is exactly my point with: #8439 
When you change this function:
`function(zephyr_library_link_libraries item)`
from:
`target_link_libraries(${ZEPHYR_CURRENT_LIBRARY} ${item} ${ARGN})`

to always do:
`target_link_libraries(${ZEPHYR_CURRENT_LIBRARY} PUBLIC ${item} ${ARGN})`

then you can not specify, e.g. anymore:
`zephyr_library_link_libraries(PRIVATE <item>)`
or
`zephyr_library_link_libraries(INTERFACE <item>)`

instead you end up having to create even more wrapper functions.

SebastianBoe, 2018-06-29:
Ah, my mistake, I thought this was a bug-report.

e.g. I thought you were reporting that

target_link_libraries(mylib INTERFACE <item>)

was not possible.

If this is a feature-request, could you rephrase from:

"cmake: Zephyr wrapped functions does not allow keywords on zephyr_link_libraries"

to

"cmake: Add zephyr_library_* wrapper support for interface and private linking of libraries"

Unless this is a bugreport, e.g. it is trying to report that 

"cmake: The zephyr_library_* wrapper does not scale"

tejlmand, 2018-07-04:
Well, I think this is a minor bug.

As part of: #8445 the flag _ConfigAbsSyms should be used when linking to libzephyr.a
 but as it is a link only flags, i.e. should not be used by the module itself, then the correct fix would be to write:
zephyr_library_link_libraries(INTERFACE -u_ConfigAbsSyms)
but as described in this issue this fails.

Your solution only targets the addition of PUBLIC / PRIVATE keywords to wrapped CMake functions.
It doesn't add the possibility of specifying the INTERFACE keyword to:
zephyr_library_link_libraries()

SebastianBoe, 2018-08-20:
Hi,

I believe that the title and/or the issue description is out-of-date.

E.g. it is stated in the description:

> During work on: #8445 I discovered that the wrapping macro: zephyr_library_named()
> break the possibility of later using PUBLIC / PRIVATE / INTERFACE keywords together with
> [...]
>  using plain cmake:
> target_link_libraries(myLib INTERFACE <linking_flag>)
> 

but after https://github.com/zephyrproject-rtos/zephyr/pull/9309 was merged this is no longer the case.

Could you either update it, or permit me to update it as I understand it?

SebastianBoe, 2019-01-16:
Closing due to the reporter not responding to the assignee.

@tejlmand : Feel free to re-open after addressing the comment from August.

closed 2019-01-16 by SebastianBoe