zephyrproject-rtos/zephyr #8626: Changes to SOCs (and ARCHs) are not covered thoroughly enough by PR CI runs

aescolar, 2018-06-28:
As shown by #8602 happening, we have a hole in the CI runs when SOCs are modified.

It would be nice if changes in SOCs would trigger (at least some of) its boards to be run in CI during its PR.

Thoughts:
I guess one option could be to have a get_modified_socs.py similar to get_modified_boards and get_modified_tests, which would add the boards based on a given SOC being run.
Another could be to have at least 1 board per SOC as testing defaults for a given arch.

nashif, 2018-12-08:
While that would be nice, it will generate a huge amount of tests to be built on PRs, some SoCs have way too many boards to cover in a single CI run for each PR. This is now covered in a daily build of all boards and SoCs.

gmarull, 2022-08-11:
is this still applicable?

aescolar, 2022-08-11:
I haven't looked recently.
@nashif can you answer?

nashif, 2022-11-20:
This and #44593 and many other areas, all the same issue, we do not have the connection between socs and boards, drivers and socs, driverers and architectures, west modules and features and hal and boards, it is all very fuzzy right now and it is not easy.