zephyrproject-rtos/zephyr #8631: memory leak in mbedtls using DTLS client

mazhenke, 2018-06-29:
I am using the DTLS client in zephyr, and I found there is memory leak when run the following flow mutli times:

1. net_app_init_udp_client
2. net_app_set_cb
3. net_app_client_tls
4. net_app_connect
5. net_app_close
6. net_app_release
7. go to step1 and repeat

Seems the mebedtls_ssl_xxx_init functions in _net_app_tls_init function doesn't release the heap when calling net_app_release

mateusz-, 2018-06-30:
I think I am also running into this. If I call mqtt_connect a couple of times I end up getting an error in my setup cert callback. Specifically, mbedtls_x509_crt_parse_der fails with MBEDTLS_ERR_X509_ALLOC_FAILED.

My callback is straight out of the mqtt example file.

mike-scott, 2018-07-04:
Hello,

> I am using the DTLS client in zephyr, and I found there is memory leak
> when run the following flow mutli times:
>
>  1. net_app_init_udp_client
>  2. net_app_set_cb
>  3. net_app_client_tls
>  4. [...]

I have a WIP patchset which includes this hack to fix the leak you're 
describing.  I need some time to submit an actual fix:
https://github.com/zephyrproject-rtos/zephyr/pull/8570/commits/82c5d82a09bcee33351a8f1ecc5aed7f490d4c63#diff-88ec0070db0157cb49aeb6609e210f1aR3883

Not sure if knowing where the leak is helps, but always good to get more 
information.

NOTE: Added Jukka for reference

- Mike

pfalcon, 2018-08-21:
Similar comment to what was posted to https://github.com/zephyrproject-rtos/zephyr/issues/8746 :

Recently, DTLS socket support was merged: https://github.com/zephyrproject-rtos/zephyr/pull/9338 , and that's going to be the new official API for (D)TLS usage. You are thus recommended to switch to it whenever possible. You may find that older net_app DTLS client is in "maintenance mode" (that actually depends on developers' time availability, and AFAIK, they're fully occupied). I thus lower priority of this ticket.

tautologyclub, 2019-01-31:
I'm guessing all net_app-dtls issues will be more or less left to hang henceforth? 

That's understandable I guess, but I'd make sure to mark it as `__deprecated` if so.

We discovered a really weird bug today where net_app registered a rx callback which was called by one of the handshake functions, but the rx callback itself indirectly called the handshake function itself - thus recursing itself (silently, for some reason) into oblivion.

Our current intuition says that we should probably forget about net_app and migrate to sockets because it seems like net_app is a lot more `__deprecated` than the documentation suggests. True?

jukkar, 2019-02-25:
net-app is gone, so closing this issue

closed 2019-02-25 by jukkar