zephyrproject-rtos/zephyr #8643: Add SAADC driver for nRF52

rodrigopex, 2018-06-29:
I have the SAADC working on my personal project using NRFX Hal, and I would like to "port" this code and making a driver for SAADC on Zephyr. Would I create a specific driver for SAADC cloning the ADC or should I put the driver as another ADC driver for nRF52 devices?

carlescufi, 2018-06-29:
Hi @rodrigopex. Once this PR is merged you should have what you need:
https://github.com/zephyrproject-rtos/zephyr/pull/7691/commits

Mind closing this issue if that covers your requirement?

rodrigopex, 2018-06-29:
Thank you. I will cover what I need.

closed 2018-06-29 by rodrigopex

rodrigopex, 2018-06-29:
@carlescufi I have already done some code for this issue. As it is almost ready based on the #7691, I will discard my code. I did a sample for the ADC and I was wondering if it would be useful as we don't have any ADC sample code. This is the link for the branch with the implementation I did: https://github.com/rodrigopex/zephyr/tree/saadc_driver. It is really simple and basic but it solves what I need. It is using the "old" ADC API.