zephyrproject-rtos/zephyr #8658: tests/net/trickle fails on FRDM k64f

inakypg, 2018-06-30:
Toolchain: ZEPHYR_GCC_VARIANT=zephyr

Commit 8fe9f432d29d7c1a5851c8808ccbc1e8a43a5438

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.99 (delayed boot 1000ms) *****
Running test suite test_tickle
===================================================================
starting test - test_init
PASS - test_init
===================================================================
starting test - test_trickle_create
PASS - test_trickle_create
===================================================================
starting test - test_trickle_start
PASS - test_trickle_start
===================================================================
starting test - test_trickle_1_status
PASS - test_trickle_1_status
===================================================================
starting test - test_trickle_2_status
PASS - test_trickle_2_status
===================================================================
starting test - test_trickle_Trickle 1 0x20000128 callback called
1_wait

    Assertion failed at tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
Trickle 1 no timeout
FAIL - test_trickle_1_wait
==============================Trickle 2 0x20000170 callback called
=====================================
starting test - test_trickle_2_wait

    Assertion failed at tests/net/trickle/src/main.c:169: test_trickle_2_wait: (cb_2_called is false)
Trickle 2 no timeout
FAIL - test_trickle_2_wait
===================================================================
starting test - test_trickle_1_update
PASS - test_trickle_1_update
===================================================================
starting test - test_trickle_2_inc
PASS - test_trickle_2_inc
===================================================================
starting test - test_trickle_1_status
PASS - test_trickle_1_status
===================================================================
starting test - test_trickle_1_wait_long
SKIP - test_trickle_1_wait_long
===================================================================
starting test - test_trickle_stop
PASS - test_trickle_stop
===================================================================
starting test - test_trickle_1_stopped
PASS - test_trickle_1_stopped
===================================================================
===================================================================
RunID: ci-180629-0939-1861:5sye
PROJECT EXECUTION FAILED
```

jukkar, 2018-07-02:
I could not reproduce this after running trickle test about 50 times in frdm-k64f board. Is this happening all the time in your setup or is it just a random failure?

inakypg, 2018-07-02:
since 6/18 I have a total of 36 runs and 20 failed, spread around multiple targets (see below).

A thought: on some of them it fails cb_2_called, in some cb_1_called. I had not noticed this before. Could this be another awkward interaction with CONFIG_BOOT_DELAY?

```
FAIL ci-180623-1215-1829:e1xu zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ pqoh on 2018-06-23 20:36:49.251000
  INFO1    will run on target group 'target=srrsotc05/frdm-03:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180623-2354-1831:rlsh zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ axkg on 2018-06-24 08:04:41.173000
  INFO1    will run on target group 'target=srrsotc04/frdm-10:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180625-0846-1835:tdzj zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ i46m on 2018-06-25 16:00:05.829000
  INFO1    will run on target group 'target=srrsotc03/frdm-01q:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180625-1314-1836:rlsh zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ axkg on 2018-06-25 20:27:44.483000
  INFO1    will run on target group 'target=srrsotc04/frdm-10:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180625-1931-1837:e1xu zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ pqoh on 2018-06-26 03:44:28.774000
  INFO1    will run on target group 'target=srrsotc05/frdm-03:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180626-1502-1843:xuxe zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ 4ju6 on 2018-06-26 22:38:51.825000
  INFO1    will run on target group 'target=jfsotc04/frdm-12k:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180626-1931-1844:tdzj zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ i46m on 2018-06-27 03:04:46.854000
  INFO1    will run on target group 'target=srrsotc03/frdm-01q:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180627-1108-1848:tdzj zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ i46m on 2018-06-27 18:54:54.879000
  INFO1    will run on target group 'target=srrsotc03/frdm-01q:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180627-1312-1849:rlsh zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ axkg on 2018-06-27 20:46:22.817000
  INFO1    will run on target group 'target=srrsotc04/frdm-10:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180627-1656-1851:xuxe zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ 4ju6 on 2018-06-28 00:38:22.353000
  INFO1    will run on target group 'target=jfsotc04/frdm-12k:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180627-1931-1852:rlsh zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ axkg on 2018-06-28 03:07:37.205000
  INFO1    will run on target group 'target=srrsotc04/frdm-10:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180628-0502-1853:rlsh zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ axkg on 2018-06-28 13:04:10.774000
  INFO1    will run on target group 'target=srrsotc04/frdm-10:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180628-0900-1855:tdzj zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ i46m on 2018-06-28 17:06:47.015000
  INFO1    will run on target group 'target=srrsotc03/frdm-01q:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
FAIL ci-180628-1115-1856:xuxe zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ 4ju6 on 2018-06-28 19:27:23.845000
[... cut ...]
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:169: test_trickle_2_wait: (cb_2_called is false)
FAIL ci-180702-0815-1869:e1xu zephyr.git/tests/net/trickle/testcase.yaml#net.trickle @ pqoh on 2018-07-02 16:35:11.436000
  INFO1    will run on target group 'target=srrsotc05/frdm-03:arm'
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
              console output Assertion failed at zephyr.git/tests/net/trickle/src/main.c:140: test_trickle_1_wait: (cb_1_called is false)
```

jukkar, 2018-07-03:
Could be related to boot delay parameter, I did not had that in my setup.

jukkar, 2018-07-04:
FYI, after setting CONFIG_BOOT_DELAY=1000, I managed to replicate the issue.

PR #8744 (2018-07-06): tests: net: trickle: Initialize test variables earlier
The cb_[1|2]_called variables were initialized just before running
the test. This was too late as in same cases the Trickle timer
was triggered before this init, which then caused the test to fail
even thou there was no actual timer failure.

This was seen with frdm_k64f and when CONFIG_BOOT_DELAY=1000 was set.

Fixes #8658
commit f5ec5674 (2018-07-06): tests: net: trickle: Initialize test variables earlier

The cb_[1|2]_called variables were initialized just before running
the test. This was too late as in some cases the Trickle timer
was triggered before the variables were init, which then caused the
test to fail even though there was no actual timer failure.

This was seen with frdm_k64f and when CONFIG_BOOT_DELAY=1000 was set.

Fixes #8658
closed 2018-07-06 by jukkar