zephyrproject-rtos/zephyr #8699: [Coverity CID :186842] Memory - illegal accesses in /drivers/interrupt_controller/plic.c

mandarcthorat1, 2018-07-03:
Static code scan issues seen in File: /drivers/interrupt_controller/plic.c
 Category: Memory - illegal accesses
 Function: plic_irq_handler
 Component: Drivers
 CID: 186842
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

agross-linaro, 2018-07-05:
It seems to me like there are some missing #idfefs around the irq +=  RISCV_MAX_GENERIC_IRQ; keying off of CONFIG_RISCV_HAS_PLIC.  However there might be issues with the PLIC_IRQS count.  With coverity thinking this only has 32 IRQS, it appears the failing platform is the one without the HAS_PLIC configured.

nashif, 2018-07-31:
@kgugala can you please have a look?

nashif, 2018-08-14:
@kgugala ping

kgugala, 2018-08-29:
@nashif sorry for stalling this, I was on holidays. I'll take a look on that.

ceolin, 2018-08-31:
guys, this is not a bug. The invalid memory access that coverity is complaining will never happen because ``_irq_spurious(NULL);`` will be called first and this function never returns

```
...
_NanoFatalErrorHandler(_NANO_ERR_SPURIOUS_INT, &_default_esf);
```

Question, should I add a return after this so any static analyzer will complain or we can just close this issue (here and in coverity)

nashif, 2018-08-31:
lets close.. mark it as invalid in coverity please.

closed 2018-08-31 by nashif