zephyrproject-rtos/zephyr #8705: Out of the box error in samples/subsys/nvs with nRF52-PCA10040

GerlichM, 2018-07-03:
Hello,

i want to use the subsys/nvs sample with the nRF52-PCA10040 board with Zephyr version 1.12
The (unedited) sample compiles fine with cmake and ninja but after flashing the following error messages occur:

***** Booting Zephyr OS 1.12.99 *****                                           
Flash Init failed                                                               
No address found, adding 192.168.1.1 at id 1                                    
No key found, adding it at id 2                                                 
No Reboot counter found, adding it at id 3                                      
Id: 4 not found, adding it                                                      
Reboot counter history: Error, no Reboot counter

[image]

For some reason it cannot initialize the nvs in
...
rc = nvs_init(&fs, FLASH_DEV_NAME, STORAGE_MAGIC);
	if (rc) {
		printk("Flash Init failed\n");
	}
...

The default configurations in prj.config are:
CONFIG_FLASH=y

CONFIG_NVS=y
CONFIG_NVS_LOG=y
CONFIG_NVS_LOG_LEVEL=4
CONFIG_REBOOT=y

Thanks!

carlescufi, 2018-07-04:
CC @Laczen

Laczen, 2018-07-04:
Hi @GerlichM , could you add the following line to the config file:
CONFIG_MPU_ALLOW_FLASH_WRITE=y

GerlichM, 2018-07-09:
Hi @Laczen ,
thank you for your answer.
I added  CONFIG_MPU_ALLOW_FLASH_WRITE=y to the config file,
the output still shows "flash init failed"

Laczen, 2018-07-09:
Hi @GerlichM, I don't have a nrf52-PCA10040 board, so it is quite difficult for me to guide you.

However I have seen that the nrf52 has a flash page size of 4096, maybe this is the reason. Could you change the sample to: NVS_SECTOR_SIZE 4096 and retry.

nashif, 2018-08-27:
not able to reproduce this with the mentioned board.

closed 2018-08-27 by nashif