zephyrproject-rtos/zephyr #8722: Network stack cleanup: connection

tbursztyka, 2018-07-04:
connection.c needs cleanup:
 . revise the connection cache logic: use sys_slist instead of looping on an array. See what has been done in subsys/net/l2/ethernet/arp.c

declutter using # rules

pfalcon, 2018-08-22:
I'd like to propose simplification of connection callback functions, let me crossreference it from here: https://github.com/zephyrproject-rtos/zephyr/issues/8723#issuecomment-415040335

pfalcon, 2018-11-07:
> @pfalcon You are taking things out of the context and making false conclusions.

The point is that there's not enough context, guidelines, and requirements are provided, so one literally can only try to make conclusions from the sparse data provided, which can easily turn out to be false...

tbursztyka, 2018-11-07:
I think we all know what needs to be done here, no need to take by hand whoever is going to do that work.

pfalcon, 2018-11-07:
> I think we all know what needs to be done here, no need to take by hand whoever is going to do that work.

So let me clarify: I don't know, so that's why I didn't participate in it. Actually, I had issues with reviewing patches claiming to be cleanups/fixes based on this and related tickets - if it's known exactly *what* should be done and *why*, it's hard to assess whether a particular change is improvement, regression, or just sideways work.

Anyway, just a friendly feedback.

commit 9a6e3012 (2019-05-07): net/connection: Final cleanup

- renaming functions to better names
- reordering functions place (register, then unregister for instance)
- centralizing logs to relevant place

Fixes #8722
commit d85aabe2 (2019-05-07): net/connection: Use slist to manage used/unused connections

This will optimize path when unused an connection is required or when
looking up a used one.

That said, at this stage, it bloats up the net_conn structure with 4
added bytes. More optimization will overcome this drawback.

Fixes #8722
commit de6fd24e (2019-05-07): net/connection: Reduce usage of #ifdef

Most of present #ifdef can be removed via using IS_ENABLED() macro.

Only small part of cache related logic still require #ifdef.

Fixes #8722
closed 2019-05-07 by jukkar