zephyrproject-rtos/zephyr #8723: Network stack cleanup

tbursztyka, 2018-07-04:
Current code base would need serious cleanup/optimization. Current status shows duplicates, un-proper modularization, under optimized data storage/lookup, naming issues etc...

What should be done:

- Keep each specific part into their own C file (i.e.: ipv4, ipv6, etc...).
  . For instance, icmpv4 redo in its code what ipv4 does already. 
- Factorize as much as possible
- Reduce #ifdefs usage inside functions
  . Use IS_ENABLED()
  . Move CONFIG_* conditioned code to own function
- Naming convention
  .When it is a public function net_\<domain\>_<...>()
  .When it is a private function \<domain\>_<...>()
- de-clutter if/for/while... imbrication. (The Linux rule of 3 maximum is nice)

Areas:

- ipv4.c/icmpv4.c : #8720
- ipv6.c/icmpv6.c : #8721
- connection.c : #8722
- rpl.c :
- net_context.c :

tbursztyka, 2018-07-04:
@jukkar, @rveerama1, @ozhuraki, @pfalcon, whoever else is interested.

pfalcon, 2018-08-22:
Here's my 2 cents of proposed cleanup: I hate these things:

~~~
tcp.c:NET_CONN_CB(tcp_established)
tcp.c:NET_CONN_CB(tcp_synack_received)
tcp.c:NET_CONN_CB(tcp_syn_rcvd)
~~~

It obfuscates the fact that it's function (so, e.g. if you search for "tcp_established(", you'll get nothing), and completely hides the function parameter list. And all this for what?:

~~~
        static enum net_verdict name(struct net_conn *conn,       \
                                     struct net_pkt *pkt,         \
                                     void *user_data)             \
        {                                                         \
                enum net_verdict result;                          \
                                                                  \
                net_context_ref(user_data);                       \
                result = _##name(conn, pkt, user_data);           \
                net_context_unref(user_data);                     \
                return result;                                    \
        }                                  
~~~

As we have 3 occurrences of this stuff, we could just expand that macro to literal code, and won't lose much. But it's actually more than that - I don't think all that wrapping is actually necessary. Just think what it does - it extends context lifetime until the end of wrapper function. But the wrapper function doesn't use context in any way after the wrappee returns, so this extending is completely useless! It only adds confusion on the source level and extra callframe on the stack.

pfalcon, 2018-08-22:
@jukkar , @tbursztyka : Can you assess the proposal in previous comment please?

pfalcon, 2018-08-22:
> It obfuscates the fact that it's function (so, e.g. if you search for "tcp_established(", you'll get nothing), and completely hides the function parameter list.

https://github.com/zephyrproject-rtos/zephyr/pull/9564 tries to alleviate at least concerns about with code comments.

pfalcon, 2018-08-22:
Overall, TCP is the most atrocious part of the stack code.  https://github.com/zephyrproject-rtos/zephyr/issues/8726 doesn't pay it good attention. I've created #9570 to discuss the real guts of it.

tbursztyka, 2019-04-17:
edited: removed the TCP cleanup tasks.

tbursztyka, 2019-07-09:
done

closed 2019-07-09 by tbursztyka