zephyrproject-rtos/zephyr #8724: Network stack cleanup: RPL

tbursztyka, 2018-07-04:
rpl.c needs cleanup:
 . instances should be managed through sys_slist, see what has been done in subsys/net/l2/ethernet/arp.c
 . revise struct net_rpl_instance: k_delayed_work is an heavy object, 1 for each timer should be enough. See again arp.c
 . If possible (I don't know rpl myself so, just wet-finger idea): once sys_slist logic is in place, try using union on some of the attributes. Like done in struct arp_entry

declutter using #8723 rules

rveerama1, 2018-10-18:
Deprecating RPL, so closing this issue.

tbursztyka, 2018-10-23:
@jfischer-phytec-iot Are you planning to maintain rpl code?

jfischer-phytec-iot, 2018-10-23:
@tbursztyka at last keep it alive

tbursztyka, 2018-10-23:
@jfischer-phytec-iot ok, because there is a lot of work in there. Not only cleaning up the code base I mean. A lot of features are still missing

laperie, 2019-02-05:
Closing since we removed RPL

closed 2019-02-05 by laperie