zephyrproject-rtos/zephyr #8730: ARM: Use a unified K-config option for enabling (default) ARM MPU support

ioannisg, 2018-07-04:
Several ARM-based SOCs provide a K-config option to allow the user to choose whether to enable the ARM MPU. Example from Beetle:

```
config ARM_MPU_ENABLE
	bool "Enable MPU on ARM Beetle"
	depends on CPU_HAS_MPU
	select ARM_MPU
	help
	  Enable MPU support on ARM Beetle SoCs
```
This option is re-defined in several SOCs
- Beetle
- Nordic nRF52
- NXP i.MX RT
- STM32 SoCs,

however, it always does the same thing: selecting ARM_MPU.

There is a proposal (by @pizi-nordic ) to define a common option in arch/arm/core/Kconfig instead of multiple duplicate options in the respective SOC definition folders:

config ARM_MPU_ENABLE
	bool "Enable the use of ARM MPU"
	depends on CPU_HAS_MPU
	select ARM_MPU
	help
	  Enable MPU on ARM SoCs that implement the standard ARM MPU.

Does this K-config simplification make sense?

ioannisg, 2018-07-04:
Copying @galak , @MaureenHelm , @erwango , @carlescufi

pizi-nordic, 2018-07-04:
I see no difference between ARM_MPU_ENABLE and ARM_MPU in proposed scheme. Could we just make ARM_MPU visible to the developer?

PR #10024 (2018-09-20): arm: Cleanup Kconfig options controlling MPU support.
Situation before this PR:
- Depending on the platform, user has to select one of the per-platform options controlling MPU
   (HAS_SYSMPU, ARM_MPU_IMX_RT, ARM_MPU_NRF52X, ARM_MPU_ENABLE or STM32_ARM_MPU_ENABLE)
- Each of these option just selected ARM_MPU or NXP_MPU, which selected ARM_CORE_MPU.
  - ARM_CORE_MPU controlled common code.
  - ARM_MPU/NXP_MPU selected which MPU driver should be compiled-in.

Situation after this PR:
- On all platforms user just needs to turn on ARM_MPU option in order to enable MPU support.
  - Correct driver is [... cut]
closed 2018-09-20 by carlescufi