zephyrproject-rtos/zephyr #8734: USB: DFU timeout on nRF52840 due to long flash erase

pawelzadrozniak, 2018-07-04:
Attempt of downloading new firmware with dfu-util on nRF52840 results in timeout due to flash erase operation taking too much time (flash_img_buffered_write() called from dfu_flash_write() in usb_dfu.c.

This problem could be solved by fragmenting whole flash erase operation into smaller page erases when needed.

Working USB driver for testing is available on this fork/branch:  https://github.com/pawelzadrozniak/zephyr/tree/nrfx_usbd3 .

Qbicz, 2019-03-14:
I faced the same issue as described here: https://lists.zephyrproject.org/g/users/topic/19099852

I worked around the timeout by compiling dfu-util with timeout of 20 seconds: https://sourceforge.net/p/dfu-util/dfu-util/ci/master/tree/src/dfu.c

Build instructions here: http://dfu-util.sourceforge.net/build.html

I was able to do the USB DFU on nRF52840 PCA10056.

I just wonder, there's an option https://github.com/JuulLabs-OSS/mcuboot/blob/master/boot/zephyr/Kconfig#L116 for the serial recovery DFU in MCUBoot. Could it be implemented also for USB DFU class in Zephyr application (background DFU)?

nvlsianpu, 2019-03-14:
fore sure it can bee
cc @carlescufi Its about "Erase flash progressively when receiving new firmwar" for zephyr.

carlescufi, 2019-03-14:
>I just wonder, there's an option JuulLabs-OSS/mcuboot:boot/zephyr/Kconfig@master#L116 for the serial recovery DFU in MCUBoot. Could it be implemented also for USB DFU class in Zephyr application (background DFU)?

So you want the SMP (Simple Management Protocol) over USB over CDC ACM? @lemrey did something similar in MCUboot itself, and this should already work in Zephyr if you enable CDC ACM and then select the serial port for `mcumgr` to be that CDC ACM port

carlescufi, 2019-03-14:
>I worked around the timeout by compiling dfu-util with timeout of 20 seconds instead of 5 seconds: 

Can you submit that fix upstream to `dfutil` so we can see it fixed in upcoming versions?

Qbicz, 2019-03-15:
>> I just wonder, there's an option JuulLabs-OSS/mcuboot:boot/zephyr/Kconfig@master#L116 for the serial recovery DFU in MCUBoot. Could it be implemented also for USB DFU class in Zephyr application (background DFU)?

> So you want the SMP (Simple Management Protocol) over USB over CDC ACM? @lemrey did something similar in MCUboot itself, and this should already work in Zephyr if you enable CDC ACM and then select the serial port for mcumgr to be that CDC ACM port

The option I mentioned is BOOT_ERASE_PROGRESSIVELY. My comment applies to storing the image received by USB DFU class. It could erase the flash progressively, when the page is needed, instead of doing it all at once and possibly hitting timeout.

> Can you submit that fix upstream to dfutil so we can see it fixed in upcoming versions?

Yes, I will definitely do that.

nvlsianpu, 2019-03-15:
If needed we can add BOOT_ERASE_PROGRESSIVELY to subsys/dfu/img_util. This will work for any DFU solution which uses this module (AFIK: USB class DFU, USB CDC ACM DFU, BLE DFU, Serial DFU).

lemrey, 2019-03-15:
Yes, the solution described by @pawelzadrozniak is how we solved the issue in MCUboot with `BOOT_ERASE_PROGRESSIVELY`. You can add me to review the patch to this repo.

nvlsianpu, 2019-03-17:
@Qbicz @lemrey WIP here https://github.com/nvlsianpu/zephyr/tree/feature/progresive_image_erase

tormodvolden, 2019-04-09:
As I explained in the pull request, the problem is not that the timeout in dfu-util is too small (the device should always respond immediately to the DFU request USB transaction) but that the device is not correctly implementing the DFU standard and the bwPollTimeout in the DFU_GETSTATUS response.

PR #15330 (2019-04-16): usb: dfu: set bwPollTimeout dynamically
Set bwPollTimeout for DFU_GETSTATUS request dynamically.
For now, adjust bwPollTimeout only during DNLOAD stage.

Fixes #8734
commit a05ef255 (2019-04-16): usb: dfu: set bwPollTimeout dynamically

Set bwPollTimeout for DFU_GETSTATUS request dynamically.
For now, adjust bwPollTimeout only during DNLOAD stage.

Fixes: #8734
PR #14662 (2019-04-28): DFU: progresive image erase
Patch adds option for progressive erase of firmware image.
Using this flash is erased as necessary when receiving
new firmware, instead of erasing the whole image slot at once.
This is necessary on some hardware that has long erase times,
to prevent long wait times at the beginning of the DFU process.

usb/class/usb_dfu: support progressive image erase

Disable bulk slot image erase when progressive erase is on.
Erase of image bank is performed by image collection procedure
progressively.

For some SoC like nRF52840 bulk erase take too much time, which
cause timeout [... cut]
closed 2019-04-16 by nashif